raspberrypi/pico-feedback #1: ADC DMA sample / documentation

taunusflieger, 2021-01-22:
It would be nice to have some additional information on how to use ADC with DMA. The current documentation feels a little bit light (not clear how to setup the DMA transfer)

Thanks Michael

Wren6991, 2021-01-24:
I had a look at that section and yes it is a bit sparse! Will take a while to get a doc update pushed through, so in the meantime I thought it might be helpful to add some example code. See https://github.com/raspberrypi/pico-examples/pull/15

Note there is some slightly more helpful information in the register listings, but I'll flesh out that section you pointed out.

Wren6991, 2021-01-26:
[image]

This will appear in the next docs release (should be imminent). If it still lacks clarity or detail then please do ask, otherwise please close if you feel the issue is resolved

aallan, 2021-01-26:
There should be a V1.1 release later today.

aallan, 2021-01-26:
V1.1 in the wild.

closed 2021-01-26 by aallan