raspberrypi/pico-feedback #4: Avoid pico-setup.sh running out of space

Hermann-SW, 2021-01-23:
This might not be easy depending on which of the packages to be installed are already installed or not. But I did run multiple times into out of space, before I freed 1GB and then setup did complete successfully.

aallan, 2021-01-26:
Warning added in today's V1.1 release about the amount of space needed. Best we can do I think.

[image]

closed 2021-01-26 by aallan

Hermann-SW, 2021-01-26:
Thanks, better than nothing.

lurch, 2021-01-26:
Part of the problem is that even `apt` itself doesn't display a warning message if there's not enough free disk space - it just continues as normal and fails badly when it's unable to write anything more to disk!

Hermann-SW, 2021-01-27:
Time for an issue against apt on that?

lurch, 2021-01-27:
That's a problem for Debian to deal with, rather than Raspberry Pi. But feel free! :grinning:

Hermann-SW, 2021-01-27:
I would -- but its not that easy.

apt on github does not allow to create issues:
https://github.com/Debian/apt

It is mirrored from here, but there a 0 issues at all:
https://salsa.debian.org/dashboard/issues?scope=all&utf8=%E2%9C%93&state=all

Strange ...

lurch, 2021-01-27:
> apt on github does not allow to create issues:
https://github.com/Debian/apt

Try this: https://bugs.debian.org/cgi-bin/pkgreport.cgi?pkg=apt