raspberrypi/pico-feedback #6: Install Pico SDK by default on Raspberry Pi OS

pxeger, 2021-01-23:
It would be great if the Pico SDK could be pre-installed and pre-compiled on Raspberry Pi OS + desktop and recommended software images to avoid having to spend an hour watching it compile and install all the dependencies in front of me.

aallan, 2021-01-23:
There is a link to the source code in Git above every example in the book. Just click on the link to go directly to the raw source code.

pxeger, 2021-01-23:
What does that have to with my question? Did you comment on the wrong issue?

JamesH65, 2021-01-23:
This was considered, and may still happen, but we are trying to keep the base Raspberry Pi OS image size down.

closed 2021-01-25 by aallan

daveythacher, 2021-02-12:
It may be possible to add a module into the apt-get system. You can do this without them.

I get the point they are making however it could be a fat image or configurable image installer. I think they have some logic for this released.