raspberrypi/pico-feedback #7: Enter programming mode via gpio and not only bootsel button

papatrexas, 2021-01-23:
Hello,

As the title suggests, this will give the option of programming the pico via a remote RaspberryPi. For 1-2 units you could piggy back on the button, but I was thinking Industrial applications and many units. 
Ty

Wren6991, 2021-01-23:
Like this?

```
#include "pico/bootrom.h"
#include "pico/stdlib.h"
#define BOOT_BUTTON  0

int main() {
    gpio_init(BOOT_BUTTON);
    gpio_pull_up(BOOT_BUTTON);
    sleep_ms(1);
    if (!gpio_get(BOOT_BUTTON))
        reset_usb_boot(0, 0);
    // Otherwise continue with program...
}
```
You can read about this function in the [online docs](https://raspberrypi.github.io/pico-sdk-doxygen/group__pico__bootrom.html#ga5d51b6163ec30a147d3445cec8f33c7c)

> As the title suggests, this will give the option of programming the pico via a remote RaspberryPi
You can do this via SWD

> For 1-2 units you could piggy back on the button, but I was thinking Industrial applications and many units.
If you're surface mount soldering Pico as a module on a larger board, BOOTSEL is available as a reflowable testpoint (see the recommended footprint in the Pico datasheet)

papatrexas, 2021-01-23:
Hello Wren.

Thank you for the reply. So if I am getting this right, the code in the first part of your reply is what I need to start with (or have it checking always the state of the bootpin) every program that goes in the pico, select a gpio from the pico, define it and connect that pin to a RaspberryPi output?

Wren6991, 2021-01-23:
Yes or put it in a library and add a constructor attribute so it's injected before `main`, like the double tap bootloader example code here: https://github.com/raspberrypi/pico-examples/blob/master/system/double_tap_usb_boot/double_tap_usb_boot.c

Then link your programs against that library.

Does it need to be a GPIO, as opposed to one of the other two interfaces that are broken out (SWD or the BOOTSEL solderable testpoint on the back)?

papatrexas, 2021-01-23:
At the moment I am just exploring options.  The main thing that drives me is the fact that you don't have actual interrupts on an rpi (there is always a random delay). So I always had to use an rpi to give just wifi capabilites to an atmega128 board and handle other stuff that are not so time critical. Yhe atmega would then handle sensors and stepper drives and air valves etc.... My problem was always how you program remotely the atmega...... easily. I was hoping that the pico would solve that problem. At the same time I don't  want to force my self to surface mount stuff or reflow solder under the pico.

rin67630, 2021-01-23:
I have many Arduino Pros 3,3V connected and programmable over GPIO.
Connect TX to RX, RX to TX and GPIO17 to RST.
and use https://github.com/TOLDOTECHNIK/avrdude-gpio-autoreset.
Works like a charm over the Arduino IDE and VNC.

Wren6991, 2021-01-24:
Let me check my understanding: you want to be able to connect a couple of wires from a Pi GPIOs to some pins on the Pico, and load code onto it?

This can be done with SWD, see chapter 5 here https://datasheets.raspberrypi.org/pico/getting_started_with_pico.pdf

[image]

lurch, 2021-01-25:
And if you're not using the UART (e.g. maybe you're running the stdio over USB) then you only need the bottom 3 wires.

aallan, 2021-01-25:
As mentioned on the forums, I think SWD solves your problem.

closed 2021-01-25 by aallan