raspberrypi/pico-feedback #12: RP2040 Datasheet: Wrong code example under Dormant section

TankMasterRL, 2021-01-24:
In the RP2040 datasheet (build-date: 2021-01-21, build-version: fcd04ef-clean) there is an incorrect code example for section "2.11.5.2. Dormant".

Instead of referring to [https://github.com/raspberrypi/pico-extras/blob/master/src/rp2_common/pico_sleep/sleep.c#L134-L155](https://github.com/raspberrypi/pico-extras/blob/master/src/rp2_common/pico_sleep/sleep.c#L134-L155) it refers to [https://github.com/raspberrypi/pico-extras/blob/master/src/rp2_common/pico_sleep/sleep.c#L106-L122](https://github.com/raspberrypi/pico-extras/blob/master/src/rp2_common/pico_sleep/sleep.c#L106-L122).

aallan, 2021-01-26:
@Wren6991 Can you take a look?

Wren6991, 2021-01-27:
Thanks for the report, looks like that section needs additional cleanup but I've just switched in the correct example as you say. Will appear in next docs release.

closed 2021-01-27 by Wren6991