raspberrypi/pico-feedback #9: Cannot set 'gdbpath' in launch.json in vscode

GregAC, 2021-01-24:
Maybe this varies between vscode versions, I'm running 1.52.1 and setting 'gdbpath' in launch.json doesn't work as asked for in 6.3 in the 'Getting Started with Raspberry Pi Pico' C/C++ guide. If you attempt to launch a debug run it complains it can't find 'arm-none-eabi-gdb' and viewing the launch.json file in vscode reports 'Property gdbpath not allowed'.

gdbpath can be set globally, by going to File -> Preferences -> Extensions -> Cortex-Debug Config -> Gdb Path -> Edit in settings.json, setting to 'gdb-multiarch' allows things to work.

GregAC, 2021-01-24:
Ah looks like this has already been reported against the pico-examples repo and they provide a fix that doesn't need a global settings change, probably the better way to do it: https://github.com/raspberrypi/pico-examples/issues/8

closed 2021-01-24 by GregAC

JamesH65, 2021-01-24:
Hi Greg!