raspberrypi/pico-feedback #13: Use more legible fonts for datasheet

WestfW, 2021-01-25:
The font used for most text in the rp2040 datasheet is very thinly stroked (or something), and difficult to read.
I've attached a screenshot showing the 2040 datasheet next to a similarly scaled output from the RPi 4 datasheet, and a similar screenshot comparing it with a microchip datasheet, both of which are much more legible.
[image]

[image]

(perhaps I am just getting old.)

aallan, 2021-01-25:
One of the considerations made when choosing the font for the datasheets was printing, the legibility when printed on paper was a significant factor in the choice, as it's possible that the books may be made available in print at some point in the future. 

If you're having problems reading the text on screen, using your PDF viewer's zoom function might be an option?

lonetech, 2021-01-25:
Unfortunately this is precisely what PDF, DVI etc are designed to be bad at - they were meant to keep exact layout and font choices. Zooming, particularly with as dense a document as this, easily leads to being unable to see the whole column, let alone page. I wound up rotating my monitor 90 degrees to be able to view a whole page large enough to read comfortably. Perhaps you could consider also publishing a more reformatting friendly format, such as ePub, HTML or ODT? 

Getting old is a normal thing, and enabling a broader audience is good, particularly when the difficulty is unrelated to the subject matter and can be provided as an additional option. 

As a workaround, Calibre can convert documents between a variety of formats, and has a rudimentary viewer too. Sadly, it does a poor job with some formatting in the RP2040 datasheet, e.g. losing all the figures and tables. Most rewrapping PDF viewers do similarly. 

Calibre is also good for keeping track of hundreds of reference manuals.

aallan, 2021-01-25:
The toolchain also generates reflowable ePub, although it's still a little buggy in places, and we're thinking about releasing that. Would that resolve your issues?

aallan, 2021-01-25:
I'll take the thumbs up as a yes. Okay, good to know. I'll generate an internal Jira for myself to look at this sometime this week. Thanks for the feedback!

lonetech, 2021-01-25:
Possibly a bit quick to close, as I'm not the submitter, nor have either of us had a chance to try the ePub version. But I think it's worth a try.

aallan, 2021-01-25:
Oops! Sorry, I'm triaging a bunch of feedback, didn't spot you weren't the OP. Reopened! 😄

descartes, 2021-01-25:
Printed the shorter docs on my colour laser (Xerox VersaLink C400) - looks good but font rather slender for a 600dpi entry level printer. Printed the C SDK docs on my black & white laser (Brother HL-5250DN) - font is somewhat pixelated and the typeface where it is in colour is even worse. Having the grey box behind is wasted toner - it doesn't help readability.

For visual access reasons alone for those with old eye balls or ones that aren't working as well, a 'fatter' font would be appropriate.

lurch, 2021-01-25:
WRT "reflow tools", note that the PDFs' licence doesn't allow redistribution of modified versions.

WestfW, 2021-01-25:
I'm comparing the RP2040 datasheet to many other data sheets that I've read, from many vendors, and it has the worst legibility.

Zooming in obviously helps, but it seems to help less than I expect.

Is it also possible that the font **color** is set to something other than "black"?  I thought I had zoomed in enough to eliminate that as a possibility, but perhaps not.  Here's a "way zoomed in" (~1inch line spacing on my 4k 29inch monitor) comparison with a Microchip datasheet...

[image]

lurch, 2021-01-26:
Maybe we should use Comic Sans :rofl:

lonetech, 2021-01-26:
> Is it also possible that the font **color** is set to something other than "black"? 

Yes. It is 80% dark gray. Figures seem to use 89% dark gray. This could be the core issue (though the font is also spindly and small).

lonetech, 2021-01-28:
Incidentally, I found that the reader I prefer on my Kobo eink tablet, koreader, supports not only reflow but even contrast adjustment and cropping, both of which help read this document. Koreader is available for many platforms but its UI may be awkward, as it's designed for e-readers. http://koreader.rocks/

rowpie, 2021-01-31:
I agree that the (PDF) documents are too difficult to read.  On screen and when printed.  It does appear that the font strokes are too thin.  I suggest a change of font is required.

WestfW, 2021-02-02:
> One of the considerations made when choosing the font for the datasheets was printing, the legibility when printed on paper was a significant factor in the choice

Assuming that the fonts/colors used are the same in the "getting started with micropython ..." book that HAS been printed, there are similar complaints there:

https://www.eevblog.com/forum/microcontrollers/raspberry-pi-pico/msg3444954/#msg3444954

> The book is written for an absolute beginner with exceptional eyesight.  Damn the font is small!  These people have no respect for old people!

swetland, 2021-02-04:
I have to agree -- as someone who spends a lot of time looking through TRMs, UMs, and Datasheets in PDF, the RP2040 stands out as unusually high quality content and the absolute worst readability of any I've dealt with (even full screen on a 30" display). Please consider 100% black for text (reducing contrast hurts) and a heavier font.  The thin vertical stems really do impact readability.

aallan, 2021-02-04:
Due to the way the PDFs are generated there are some formatting issues that we're keeping track of internally. We're also going to explore some options around fonts. I'm hoping to release some reflowable ePub versions of the documentation either this week or next, which might be useful for folks. We're absolutely going to look at this, but right now we're still very close of launch and there is a lot of feedback around content to get through. Due to toolchain issues just switching out the font isn't necessarily a simple matter.

HeadBoffin, 2021-02-04:
> Due to toolchain issues just switching out the font isn't necessarily a simple matter.

Huh - what witch craft is this?

As suggested on the forum, change control like Word would be useful - or strike throughs - so we can see what's changed - with searchable text like "Here be changes". Only until things settle down.

aallan, 2021-02-10:
After some internal discussion we're going to be updating the fonts as you can see below when the next PDF documentation release happens, probably at the end of next week.

[image]

closed 2021-02-10 by aallan

aallan, 2021-02-24:
Updated PDFs are now available on the [Getting Started](https://www.raspberrypi.org/documentation/rp2040/getting-started/) pages.

swetland, 2021-02-24:
Thanks!  The new docs are significantly more readable.