raspberrypi/pico-feedback #17: Getting started with Pico pdf - Windows setup general observations

descartes, 2021-01-25:
I'm a macOS / Ubuntu / Raspberry Pi OS user normally but have clients with Windows machines so after some comments on the forum (various places) I've tried out the Windows install so I can pre-empt any support requests. Here are some notes:

8.2.1 - not obvious these are links - perhaps add "download"
8.2.1.3 - as I discovered, you can't save 3Gb of disk space by turning off "Windows 10 SDK" - note saying you need to install as is, don't try to save space.
8.2.1.5 - Installing Git is being hit with a rubber mallet of options - a check list / bullet points to make the requirements easier to read would be good.
8.2.2 - Install in to Downloads? This looks wrong, mostly because it is. Perhaps install in to a sub folder of Documents as an example but point out that you can put it anywhere you like.
8.2.2 - the GitHub addresses are clickable - we need to be able to copy & paste them - please remove the click, it gets in the way.
8.2.3 - setting the path relative means people will be tripped up if they don't create all their projects in a directory next to the SDK folder. Perhaps using the install folder from 8.2.2 you use a hard coded path? Or highlight the relative folder issue.
8.2.3 - tell them to get a cup of xxxx whilst it compiles - the first two compiles of the exe's make you think 100% is quick, not so much on the examples ;-)
8.2.4 - The blue Visual Studio Code is a link to downloads - indicate that
8.2.4 - Despite the warning, people have been tripped up with running code from the Developer Command prompt - this is so important the warning should be before and after the command prompt example.

and then the well known CMake Tools -> CMake Generator -> change this to NMake Makefiles.

aallan, 2021-01-25:
Thanks that's helpful. 

The Ninja and NMake Makefiles problem was a change to CMake Plugin (I think) that happened after the instructions were written.

descartes, 2021-01-25:
> The Ninja and NMake Makefiles problem was a change to CMake Plugin (I think) that happened after the instructions were written.

C'est la vie!

I've posted some videos for those that struggle with the install and want video assist / proof of life: https://www.raspberrypi.org/forums/viewtopic.php?f=144&t=300189

aallan, 2021-01-26:
Just had a V1.1 documentation release, which should at least document the Ninja problem. The rest is in the queue for a V1.2 release.

lurch, 2021-02-08:
> 8.2.1 - not obvious these are links - perhaps add "download"
8.2.2 - the GitHub addresses are clickable - we need to be able to copy & paste them - please remove the click, it gets in the way.
8.2.4 - The blue Visual Studio Code is a link to downloads - indicate that

Isn't it fairly common-practice for links to be show in blue text? I believe in most PDF viewers mouse-hovering over the link shows you the actual URL, and you can also either left-click it to open it in a browser, or right-click it and select "Copy Link to clipboard" ?

> 8.2.1.3 - as I discovered, you can't save 3Gb of disk space by turning off "Windows 10 SDK" - note saying you need to install as is, don't try to save space.

This is already mentioned in the V1.2 documentation release.

aallan, 2021-04-21:
I think we're pointing people towards the third-party [Pico Setup for Windows](https://github.com/ndabas/pico-setup-windows) now from the documentation. This solves most things on Windows initial setup.

closed 2021-04-21 by aallan