raspberrypi/pico-feedback #20: pico-setup.sh should not install new cmake that does not meet 3.12 requirement, when installed version suffices

Hermann-SW, 2021-01-26:
Happened on Intel Ubuntu 18, see:
https://www.raspberrypi.org/forums/viewtopic.php?f=145&t=300333&p=1804853#p1804853

lurch, 2021-02-17:
[pico-setup.sh](https://github.com/raspberrypi/pico-setup/blob/master/pico_setup.sh) was specifically written to run on Raspberry Pi OS running on a RPi, and hasn't been tested on any other platforms. (It even warns "Not running on a Raspberry Pi. Use at your own risk!" :wink: )

Would adding a `SKIP_CMAKE` option to the script resolve your problem?

Hermann-SW, 2021-02-19:
> Would adding a SKIP_CMAKE option to the script resolve your problem?
>
"cmake install" should be skipped if installed already.
So better a test whether cmake is installed, and if so, whether version installed is already sufficient for Pico.
The forum posting workaround undos the installation, whereas the new installation should better be avoided in the first place.

lurch, 2021-02-19:
> So better a test whether cmake is installed, and if so, whether version installed is already sufficient for Pico.

IMHO that sounds like far too complex logic for the relatively-simple `pico_setup.sh` bash script.

I think your particular scenario (with your usage of snappy on Ubuntu 18.04 to update cmake) is a bit of an edge-case...
Ordinarily, the version in the repos would be new-enough, and we use `sudo apt install cmake` which would update your system-installed version of cmake to whatever the latest one in the repos is. And in "normal circumstances" if your system-installed cmake package is out of date, you'd either download and extract a prebuilt binary, or you'd build and install it from source. Both of these scenarios would leave the system-installed cmake version as-is, and simply place the newer cmake earlier in your `$PATH`, so that calling `cmake` from the command-line would call your newer locally-installed version in preference to the older system-installed version.
It seems to be a bit of a "packaging bug" that both APT and snappy overwrite the same system-installed cmake binary.

So if adding a `SKIP_CMAKE` option to `pico_setup.sh` doesn't suit your purposes, I'd be tempted (pending feedback from @aallan ) to close this as "wontfix". Since as already mentioned: "Not running on a Raspberry Pi. Use at your own risk!"

aallan, 2021-02-19:
We're only promising the script works on a fully up to date Raspberry Pi OS.

Hermann-SW, 2021-02-20:
Thanks to the comments, and I agree that we should close this issue as wontfix because you provided a good solution allowing to use pico-setup.sh unchanged on Ubuntu 18 LTS:
> and simply place the newer cmake earlier in your $PATH, so that calling cmake from the command-line 
> would call your newer locally-installed version in preference to the older system-installed version.

closed 2021-02-20 by aallan