raspberrypi/pico-feedback #24: gettimeofday() does not work

Hermann-SW, 2021-01-27:
struct timeval seconds are always 0, regardless how much time passed (also for sleep_ms(3333)):
```
tud_cdc_connected()                                    
0 / 536877456                                          
0 / 3758157056                                         
0 / 536873932

```

Code:
```
#include <stdio.h>
#include "pico/stdlib.h"
#include <tusb.h>
#include <sys/time.h>

int main(void)
{
  stdio_init_all();
  while (!tud_cdc_connected()) { sleep_ms(100);  }
  printf("tud_cdc_connected()\n");

  struct timeval tv0,tv1,tv2;
  gettimeofday(&tv0, NULL);
  sleep_ms(1);
  gettimeofday(&tv1, NULL);
  sleep_ms(2);
  gettimeofday(&tv2, NULL);

  printf("%ld / %u\n",tv0.tv_sec,tv0.tv_usec);
  printf("%ld / %u\n",tv1.tv_sec,tv1.tv_usec);
  printf("%ld / %u\n",tv2.tv_sec,tv2.tv_usec);

  return 0;
}

```

dpgeorge, 2021-01-27:
You may need to initialise the RTC first, eg `rtc_init()` and `rtc_set_datetime(...)`.

Hermann-SW, 2021-01-27:
Thanks, I tried hello_rtc() demo and it worked.

I cannot reopen this issue, but it really should be reopened.
gettimeofday() does not work, see below.
If that will not be fixed for Pico, then a compile time error for its use is needed, stating to use rtc_get_datetime() instead.

```
Press CTRL-A Z for help on special keys

0 / 536873932
0 / 3758157056
0 / 536877456

```

```
#include <stdio.h>
#include "hardware/rtc.h"
#include "pico/stdlib.h"
#include "pico/util/datetime.h"
#include <sys/time.h>

int main() {
    stdio_init_all();
    sleep_ms(5000);

    rtc_init();

    struct timeval tv0,tv1,tv2;
    gettimeofday(&tv0, NULL);
    sleep_ms(3333);
    gettimeofday(&tv1, NULL);
    sleep_ms(2);
    gettimeofday(&tv2, NULL);

    printf("%ld / %u\n",tv0.tv_sec,tv0.tv_usec);
    printf("%ld / %u\n",tv1.tv_sec,tv1.tv_usec);
    printf("%ld / %u\n",tv2.tv_sec,tv2.tv_usec);

    return 0;
}

```

lurch, 2021-01-27:
ping @Wren6991

aallan, 2021-01-27:
Ah, this should have been raised against the pico-sdk repo, https://www.raspberrypi.org/forums/viewtopic.php?f=145&t=300265. Transferring the issue.