raspberrypi/pico-feedback #25: missing step in installing openocd on windows

benevpi, 2021-01-27:
On Page 55 of the Getting Started with Pico guide, the steps are:

```
$ cd ~/pico
$ git clone https://github.com/raspberrypi/openocd.git --branch picoprobe --depth=1
$ cd openocd
$ ./bootstrap
```

However, we haven't made ~/pico yet. It could either change to add the line:

```
$mkdir ~/pico
```

or just cd to the home directory since the following steps create their own sub directories anyway.

aallan, 2021-01-27:
I think the assumption at this point is that you've already worked through Chapter 2, where the `pico` directory is created to install the SDK and examples?

benevpi, 2021-01-27:
Sorry, I wasn't clear, this is building OpenOCD on Windows. 

Windows installation (chapter 8) doesn't create ~/pico. It places things in Downloads/pico-sdk and Downloads/pico-examples

aallan, 2021-01-27:
Ah, I see!

lurch, 2021-01-27:
Thanks. Fixed in the upstream version by just removing the `cd ~/pico` line :slightly_smiling_face:

closed 2021-01-27 by lurch