raspberrypi/pico-feedback #26: Errors in pico-setup.sh

BruceMardle, 2021-01-27:
In https://www.raspberrypi.org/documentation/pico/getting-started/ under the C/C++ tab, section "SDK setup", there's a button "Download Setup Script". I did that, ran the script and got errors on lines 29 and 35 ([[ not found) and 72 (Bad substitution). I don't know what (if anything) that breaks, but it probably needs fixing.

aallan, 2021-01-27:
What platform did you run the script on? It's intended to be used on Raspberry Pi running a fully updated Raspberry Pi OS.

BruceMardle, 2021-01-27:
Raspberry Pi 4 with up-to-date Raspberry Pi OS.

lurch, 2021-01-27:
Did you make the script executable with `chmod +x pico_setup.sh` ?

BruceMardle, 2021-01-27:
Yes.

aallan, 2021-01-27:
The sort of error you're getting looks like you might be trying to execute the script not in bash. Do you have a non-default shell?

BruceMardle, 2021-01-27:
Nope.
It's a newish setup, so maybe [[ lives in a package that's not installed by default?

aallan, 2021-01-27:
Are you running on Lite?

lurch, 2021-01-27:
What does `/bin/bash --version` say?

BruceMardle, 2021-01-27:
Ermm, I'm not sure. It started life as 2021-01-11-raspios-buster-armhf.zip (1,227,921,990 bytes), which didn't include the 'recommended' software (though I've installed some of it since).

lurch, 2021-01-27:
@BruceMardle I also ran pico_setup.sh on a fresh install of 2021-01-11-raspios-buster-armhf yesterday and didn't get any errors, so after some internal discussions we have got no idea what might be going wrong for you :confused: Are you able to reflash your SD card and start again afresh?

BruceMardle, 2021-01-27:
Sorry for the delay.

$ /bin/bash --version
GNU bash, version 5.0.3(1)-release (arm-unknown-linux-gnueabihf)
(etc.)

I'm only vaguely interested in the Pico. (I've already got a PIC project and an FPGA project "on the go".) I think I'll wait till other people have sorted out the early bugs!

On Wednesday, 27 January 2021, 13:24:13 GMT, Andrew Scheller <notifications@github.com> wrote: 

What does /bin/bash --version say?

aallan, 2021-01-27:
This problem hasn't been reported by any one else. 

We're happy to help with trying to diagnose your issue, but it's likely caused by local setup changes on your Raspberry Pi. To isolate your problem it the best course at this point would be to make a fresh installation of the OS on a new SD Card, and install on a clean system. Then try again.

closed 2021-01-27 by aallan