raspberrypi/pico-feedback #34: C SDK - 4.1.10. hardware_i2c - Example output incorrect

rbirkby, 2021-01-30:
raspberry-pi-pico-c-sdk.pdf section 4.1.10. hardware_i2c says:

```
   4 // I2C Bus Scan
   5 //  0 1 2 3 4 5 6 7 8 9 A B C D E F
   6 // 0
   7 // 1    @
   8 // 2
   9 // 3        @
  10 // 4
  11 // 5
  12 // 6
  13 // 7
  14 //
  15 // E.g. if slave addresses 0x12 and 0x34 were acknowledged.
```

This is incorrect. Actual output of the sample code is formatted differently. The vertical axis shows all significant digits of the hex value and empty cells are represented with a period (.) as shown below taken from actual output:

```
// I2C Bus Scan
//  0 1 2 3 4 5 6 7 8 9 A B C D E F
// 00 . . . . . . . . . . . . . . . .
// 10 . . . . . . . . . . . . . . . .
// 20 . . . . . . . . . . . . . . . .
// 30 . . . . . . . . . . . . . . . .
// 40 . . . . . . . . . . . . . . . .
// 50 . . . . . . . . . . @ . . . . .
// 60 . . . . . . . . . . . . . . . .
// 70 . . . . . . . . . . . . . . . .
// Done.
```

Additionally there is a formatting bug in the code - there is a problem with the actual output as shown above: The cells do not align with the horizontal axis because the horizontal axis needs extra padding.

lurch, 2021-02-17:
It's possible that the comment was written during an earlier version of the code, and then the code was later updated without the comment being updated to match :wink:   I'll check and update the comment as appropriate.
It looks like https://github.com/raspberrypi/pico-examples/blob/master/i2c/bus_scan/bus_scan.c is the ultimate "upstream version" of that code.

aallan, 2021-05-25:
@lurch Did the code get updated?

lurch, 2021-05-25:
Nope, not yet. Still in backlog.

aallan, 2024-01-18:
PR opened raspberrypi/pico-examples#464

closed 2024-01-18 by aallan