raspberrypi/pico-feedback #37: Document electrical limits

therealprof, 2021-01-30:
Usually MCUs have maximum sinking/sourcing capabilities per pin/per port (or group) and in total, however I can't find any such information in the datasheet.

lurch, 2021-01-30:
Does "Chapter 5. Electrical and Mechanical" contain the information you're looking for?

therealprof, 2021-01-31:
I don't see it. It talks about drive strength and voltage levels but I still have no idea what the total possible amounts of sinking/sourcing are, e.g. can could I use all GPIOs at maximum drive strength to sink or source simultaneously?

aallan, 2021-02-01:
It's a good question, and no you're right it's not information that is present in the datasheet. I'll see if I can find you an answer.

aallan, 2021-02-08:
So this turned out to be a hard question. There are a lot of different things factoring in with different constraints. The answer that the team has come to is a limit of 50mA.

[image]

The upstream documentation has been updated and this will be present in the next PDF release.

therealprof, 2021-02-08:
Allow me to follow up then. 😅 Is this for sinking and/or sourcing of currents? Possibly needs to be a sum of absolute values?

daveythacher, 2021-02-10:
Overall the Electrical Characteristics are very casual for the RP2040. It is scaled back from traditional datasheets.

A lot of this depends on how the IC is designed. There is usually a bank current limit and a global current limit. The IC will respond to differently at different temperatures, clocks, voltages, etc. 

Manufacturing process is known to affect things also. It can shift/flatten curves in a deceptive manner. Usually this is ignored.

Based on the previous comment. It would appear this is running a single bank topology which will lower cost. This is a low limit for a bank and a lot of pins for a single bank. The did provide a nice full cross bar, FIFO for IPC, etc. These are more expensive options, but were also cost optimized a little against performance. An app note could fix this. 

Honestly these are why I am interested in the Pico. The cross bar and FIFO add a few options. There are other controllers which provide similar or different notions, but in some cases these are missing completely. Raspberry Pi favors software, which is hit or miss with me. I am curious to play around with PIO, which will never replace FPGA. However it may prove interesting.

I was not very happy with how the presented the current rise against temperature and frequency. "Hot" vs "Room Temp" vs "Cold" is very casual. Now I probably could guess what they mean given my background. This was likely made as an effort to simplify things. An app note could fix this.

Note this chip could suffer from high leakage power, which while the chip is fairly power efficient it could struggle in some battery applications. Granted this is kind of getting into areas where the PICO is probably outclassed anyhow.

daveythacher, 2021-02-10:
> Allow me to follow up then. 😅 Is this for sinking and/or sourcing of currents? Possibly needs to be a sum of absolute values?

This IOVDD is likely source only. There is a max ground current somewhere. That could be the total current of logic plus IO. The pin has max current. Do not exceed the bank source limit, ground sink limit or pin source/sink limit current.

Again this depends on the topology they used. I have seen controllers with global IO limits, individual bank limits, and individual pin limits. FPGAs are known to use individual IO source pins per bank. Microcontrollers will logically divide banks internally but will tie source and sink together internally. However there is individual bank drivers internally. 

Suggestion is this uses a single bank internally.

therealprof, 2021-02-10:
@daveythacher Appreciate your efforts. I'm aware of those aspects and would agree with your speculation but I'm trying to point out deficiencies in the datasheet in the hopes that we can get some more official statement properly documented.

JamesH65, 2021-02-10:
We are looking in to it, some clarifications have already been made upstream.

daveythacher, 2021-02-10:
> @daveythacher Appreciate your efforts. I'm aware of those aspects and would agree with your speculation but I'm trying to point out deficiencies in the datasheet in the hopes that we can get some more official statement properly documented.

- Max power consumption is not specified.
- Thermal resistance is not specified.
- Internal chip logic power consumption is not specified.
- Pin timing, not specified.
- Built in interfaces timing, not specified. 
 
All kinds of different version of this exist.

daveythacher, 2021-02-10:
If you are really concerned you can just buffer the external pins. This would solve all of the problems.

therealprof, 2021-02-11:
> If you are really concerned you can just buffer the external pins. This would solve all of the problems. You would need to use CMOS or OP AMP.

I am not at all "concerned". If the Raspberry Pi chips want to be taken seriously this is an important specification item since some people select the chips for their designs by power sourcing/sinking capability and even for the amateur designer it would be nice to know if a few LEDs can be safely hooked up without the chip possibly kicking the bucket 2 weeks later, even if it is just a meager 50mA in total (NB: The hobbiest go-to ATMega328 can sink/source 40mA per pin and 200mA in total).

Expecting users to work around unknown limitations "just in case" is somewhat unreasonable. I'd love to see those MCUs succeed, that's why I'm pointing those documentation deficiencies out.

JamesH65, 2021-02-11:
And I have already stated above that we are looking in to it, and the upstream docs have already been updated.

daveythacher, 2021-02-11:
The Pi had 50mA limit on each bank also if I am not mistaken. That was only really documented on the compute modules and PI 4 right? The PICO is targeting a release structure like the compute modules right?

There is a lot to MCUs and comparing the RP2040 against AVR is just a huge sub section of that. MCU success is tied to so many things. The PICO will likely be just fine in the hobby class. There was a few short comings which the PICO may try to address.

JamesH65, 2021-02-12:
> The Pi had 50mA limit on each bank also if I am not mistaken. That was only really documented on the compute modules and PI 4 right? The PICO is targeting a release structure like the compute modules right?

GPIO on Pico was not designed by the same people who designed it on the Brcm devices used on the Pi range.

Pico and CM do not share the same release structure.

> 
> There is a lot to MCUs and comparing the RP2040 against AVR is just a huge sub section of that. MCU success is tied to so many things. The PICO will likely be just fine in the hobby class. There was a few short comings which the PICO may try to address.
> 

Time will tell, but I expect Pico to be popular in commercial devices as well as the hobby space. 

Anyway, this is a feedback repo, and there not a lot of valid feedback here. These things are better discussed on the forums, not on github.

daveythacher, 2021-02-12:
> GPIO on Pico was not designed by the same people who designed it on the Brcm devices used on the Pi range.

Missed the point.

> Time will tell, but I expect Pico to be popular in commercial devices as well as the hobby space.

I am not sure about this. For a number of reasons.

> Anyway, this is a feedback repo, and there not a lot of valid feedback here. These things are better discussed on the forums, not on github.

@JamesH65 others have said this, but I think I should repeat it. You need to calm down.

JamesH65, 2021-02-12:
Odd, feeling quite chilled and calm here. Please keep it civil on here, that means keeping away from personal attacks. And again, this sort of thing is what the forums are for. Please use them.

ghollingworth, 2021-02-12:
Removed discussion so we can focus on the issue

mike-stimson, 2021-02-22:
The datasheet will be updated with the following clarification...
[image]

therealprof, 2021-02-22:
@mike-stimson Excellent, thank you!

closed 2021-02-22 by aallan