raspberrypi/pico-feedback #39: Document maximum SPI speed

therealprof, 2021-01-30:
It's not quite clear what the maximum SPI rate is, the datasheet contains a rather confusing sentence 

> To generate a maximum bit rate of 1.8432Mbps in the master mode

but it sounds more like an example rather than the real limit. Also it's not quite clear what `PCLK` is supposed to be, probably `clk_peri` but why is the nomenclature different throughout the datasheet? That makes it hard to find the corresponding registers.

Wren6991, 2021-02-04:
Thanks, this is vestigial from the Arm documentation for the PL022 which we imported, and as you say is needlessly confusing. The maximum SPI frequency is `clk_peri` / 2. `clk_peri` is attached to `clk_sys` by default, but you can attach it to another source (e.g. directly to a PLL, with a divider) so that SPI frequency can stay constant whilst system frequency scales.

The default system frequency in the SDK is 125 MHz, so you can achieve a 62.5 MHz SCK frequency, though your actual throughput will be slightly lower due to the inter-frame gaps in the Motorola SPI mode we use. You can get the full 62.5 Mbps easily with PIO, if you check the `pio/st7789_lcd` example.

I'll update the datasheet to make sure it includes this information and is not misleading.

Wren6991, 2021-02-04:
There is this note at the top of the SPI section that gives the relationship between our system-level clocks (`clk_peri`, `clk_sys`) and the clock inputs shown on the SPI diagram (SSPCLK, PCLK) but it's worth calling this out a few more times as it's easy to miss.

[image]

therealprof, 2021-02-04:
Isn't the official max `clk_sys` 133MHz though? 😅

Wren6991, 2021-02-04:
Yes, whatever you have your `clk_sys` set to, you can get half that frequency on `SCK` without messing around with the `clk_peri` configuration.

Hexxeh, 2021-02-04:
> but you can attach it to another source (e.g. directly to a PLL, with a divider) so that SPI frequency can stay constant whilst system frequency scales

It'd be great to see an example of how to do this, since it's been oft quoted that the Pico can be heavily overclocked (and indeed it can!) but in the common way I've seen it done and tried it, it silently drops the SPI clocks down. I've managed to run at 288mhz but the SPI clock drops to around 1/3rd of the default at this point offsetting a lot of the gain.

Wren6991, 2021-02-04:
> It'd be great to see an example of how to do this

Could you please open an issue for that? We're not going to add overclocking examples to `pico-examples` since it's off-label, but we can show how to attach `clk_peri` to e.g. a divided `pll_sys` output. By default we attach it straight to `pll_usb` if you mess with `pll_sys`, which is why it gets dropped to 48 MHz.

Wren6991, 2021-02-04:
I haven't touched the existing documentation since it seems consistent and correct to me, but there are now some NOTE boxes that add context for RP2040.

[image]

[image]

closed 2021-02-04 by Wren6991

therealprof, 2021-02-04:
Excellent, thanks!

Hexxeh, 2021-02-04:
> Could you please open an issue for that? We're not going to add overclocking examples to `pico-examples` since it's off-label, but we can show how to attach `clk_peri` to e.g. a divided `pll_sys` [...]

Sure! Filed https://github.com/raspberrypi/pico-feedback/issues/63, hopefully I framed the request correctly. Thank you!

Reneg973, 2021-05-01:
Unfortunately this docu is more confusing as it doesn't tell the truth. As a colleague said: Source code is truth....
With that said, and checking the source code, - which didn't change since initial release in january, this is the code piece:
```
    // Configure PLLs
    //                   REF     FBDIV VCO            POSTDIV
    // PLL SYS: 12 / 1 = 12MHz * 125 = 1500MHZ / 6 / 2 = 125MHz
    // PLL USB: 12 / 1 = 12MHz * 40  = 480 MHz / 5 / 2 =  48MHz
    /// \end::pll_settings[]

    /// \tag::pll_init[]
    pll_init(pll_sys, 1, 1500 * MHZ, 6, 2);
    pll_init(pll_usb, 1, 480 * MHZ, 5, 2);
    /// \end::pll_init[]

    // Configure clocks
    // CLK_REF = XOSC (12MHz) / 1 = 12MHz
    clock_configure(clk_ref,
                    CLOCKS_CLK_REF_CTRL_SRC_VALUE_XOSC_CLKSRC,
                    0, // No aux mux
                    12 * MHZ,
                    12 * MHZ);

    /// \tag::configure_clk_sys[]
    // CLK SYS = PLL SYS (125MHz) / 1 = 125MHz
    clock_configure(clk_sys,
                    CLOCKS_CLK_SYS_CTRL_SRC_VALUE_CLKSRC_CLK_SYS_AUX,
                    CLOCKS_CLK_SYS_CTRL_AUXSRC_VALUE_CLKSRC_PLL_SYS,
                    125 * MHZ,
                    125 * MHZ);
```

133MHz can nowhere be found and therefore spi output is 62.5MHz.

therealprof, 2021-05-01:
@Reneg973 I have a hard time following what you're rambling about. Your code (for whatever) reason is using 125MHz as `clk_peri` so the maximum speed is indeed 62.5MHz, if the speed was 133MHz it would be 66.5MHz... coincidentally exactly as mentioned above in https://github.com/raspberrypi/pico-feedback/issues/39#issuecomment-773397615.

WestfW, 2021-05-31:
It looks like the register description for the SPI clock divisor is missing some formatting, or special characters:
[image]

lurch, 2021-06-04:
@WestfW That section of the docs was auto-converted from ARM's PL022 documentation. So see https://developer.arm.com/documentation/ddi0194/latest/ for the "upstream version" :wink: 
Specifically [here](https://developer.arm.com/documentation/ddi0194/h/programmer-s-model/register-descriptions/control-register-0--sspcr0) for SSPCR0.

WestfW, 2021-06-04:
Ah.  The autoconversion seems to have failed particularly badly on the (circled) math-like part:
[image]

aykevl, 2023-04-02:
> The default system frequency in the SDK is 125 MHz, so you can achieve a 62.5 MHz SCK frequency, though your actual throughput will be slightly lower due to the inter-frame gaps in the Motorola SPI [...]

I've found these inter-frame gaps to reduce SPI speed quite substantially.

For anybody trying to get the highest possible SPI speed without going all the way to PIO, for me I could get the closest to the maximum speed using SPI mode 3 and by using DMA instead of regular register writes. With my measurements, that gets me around 8.5 clock cycles per byte. (Why not 8 or 9? I don't know, my measurement could be wrong. But it's very close to 8.5 clock ticks).
I suppose using 16-bit SPI might reduce the overhead even further.