raspberrypi/pico-feedback #41: Preinstall uPython UF2

PiGraham, 2021-01-31:
Not a big thing, but it would be an easier intro to new users if they could install Thonny and go straight into REPL and USB upload of code right from the start.

That might also remove possible confusion about the need to press BOOTSEL to load code.

lurch, 2021-01-31:
I _suspect_ that we won't be changing that at this stage (although I guess there's no reason that a 3rd-party reseller couldn't sell boards with MicroPython pre-flashed for an additional fee?).

Are the [existing instructions](https://www.raspberrypi.org/documentation/pico/getting-started/) not clear enough? Anything we could do to make them easier to follow?

PiGraham, 2021-01-31:
Noted.

Regarding instructions, they do work work.

> You can program your Pico by connecting it to a computer via USB, then dragging and dropping a file onto it, so we’ve put together a downloadable UF2 file to let you install MicroPython more easily.

This could give the impression that BOOTSEL and drag and drop is how you program a Pico. This is how some other MCU boards are programmed (CodeBug, MicroBit), but this is not the case with uPython. Indeed you can not put your uPython code on a Pico this way.  This could be a little misleading.

JamesH65, 2021-01-31:
That's a good point, that might need some extra explanation.

closed 2021-02-01 by aallan