raspberrypi/pico-feedback #42: Code upload over USB

PiGraham, 2021-01-31:
It would be more convenient for may users to upload code over USB without requiring BOOTSEL & UF2 file. 
This is provided on Pico with uPthon / Thonny. Other MCU boards allow such a method.

SWD provides more functionality, but it is an additional interface. It requires new users to solder to their Pico, have programmer hardware (Pi, Pico, other)  and there will be many use cases where the uUSB is accessible but the debug port is not.

Hello world running in <5 mins with Thonny but >30 mins to do the same in C.

It could enhance the new (maybe very experienced) user's out of the box experience.

lurch, 2021-01-31:
Interesting idea (which I guess would be _theoretically_ possible using some kind of "bootloader" UF2 file which offers a MSD interface over USB, to which new code could be written). But if you're e.g. using the USB interface for stdio (or some other USB task like HID or audio), how would you tell the board to re-enter the "code upload over USB" process?

The advantage of the BOOTSEL button is that it *always* works, and there's no way of making it "not work" (as it's in the chip's ROM) :slightly_smiling_face:

PiGraham, 2021-01-31:
I don't know the details, but it works on Pico with uPython and it works on various Arduino and ESP LUA boards.
I think it adds a short delay on boot that the bootloader and uploader can detect to stop user code and enable upload.
It doesn't require MSD. I think Arduino does it over USB serial.

Certainly keep the BOOTSEL. Espressive boards (ESP32/ESP8266) so something similar but use a GPIO0 tied low at boot to force bootload.
Arduino allow reprogramming a corrupted bootloader via a programming header that works with e.g. FTDI USB Serial, or an onboard USB Serial chip.

ghollingworth, 2021-01-31:
I agree with others that this is something that should happen when Arduino release their RP2040 derivative.  So there's no point in us doing it

closed 2021-01-31 by ghollingworth