raspberrypi/pico-feedback #43: Better Hello World experience for C

PiGraham, 2021-01-31:
A suggestion.

It might be a bit more rewarding if the C SDK Hello World examples were the typical type some code, compile, upload & run rather than the drag & drop of pre-compiled UF2 files. I think it would give more of a "I made this" experience than an "I ran this App" experience.

That would probably rely on a simpler start more like you have with Thonny uPython. Type, save, run & just the USB connection.
So some changes would be needed to support that.

lurch, 2021-01-31:
We go into this more in-depth in "Chapter 7. Creating your own Project" of https://datasheets.raspberrypi.org/pico/getting-started-with-pico.pdf - which you'd hopefully agree is a bit too complicated for a "very first step with the Raspberry Pi Pico" ? :shrug:

PiGraham, 2021-01-31:
> a bit too complicated for a "very first step

Yes, but maybe there is some streamlining that could be done there.
Could you apt install a ready to run IDE?
In conjunction with USB upload it should be possible to New...type a few lines, click build, upload and its done.
Is there anything fundamental preventing C Hello World not being as easy as uPython hellow World?

lurch, 2021-01-31:
Perhaps what you're asking for will be possible in future when IDEs develop "more streamlined" support for the Raspberry Pi Pico? But IMHO that's something for the IDE developers to tackle, rather than Raspberry Pi Trading. (but that's just my personal viewpoint rather than any "official policy"!)

JamesH65, 2021-01-31:
We have been looking in to a "single click installation" for VScode and all the ancillaries. The pico_install script should do most of that of course, but we need something for other platforms. There certainly some niggles to iron out.

PiGraham, 2021-01-31:
> Perhaps what you're asking for will be possible in future when IDEs develop "more streamlined" support for the Raspberry Pi Pico? But IMHO that's something for the IDE developers to tackle, rather than Raspberry Pi Trading. (but that's just my personal viewpoint rather than any "official policy"!)

I don't think it really an IDE issue at all. It's a basic data transfer function. It should be as applicable to command line. You can do that on Arduino, ESP etc. Fundamentally it needs some bootloader code on the Pico, as initial entry point 'user' code that can receive program code and stuff it into flash or find no uploader present on the link  and run next level user code.
There are plenty of precedents for this. Pico uPython, Arduino etc...

ghollingworth, 2021-01-31:
Since this ends up being "make it easier" then I'm not sure there's anything here to discuss or change.  There are lots of third party contributors already looking into developing and extending their tools to support RP2040, I think this is something we'll just have to watch as time goes on (and take it into consideration for any future tutorial releases).

We should probably keep an eye out in the long term how people use the tutorials and getting started guides but I don't think there's anything to change right now.

closed 2021-01-31 by ghollingworth