raspberrypi/pico-feedback #44: Python serial GPIO control for beginners

FunkyGandalf, 2021-01-31:
Loving the RPi 4 and learning robotics through it, there has been enough excellent documentation for me to build a fairly complex (for me) little robot I'm now looking to expand by using a Pico to give me more gpio pins and better servo control.  

I would love to see more documentation for beginners at how to use serial communication within a python script to control hardware on the Pico.  Most of the beginner documentation I see uses scripts run on the Pico itself, but I want to use a Pi for the heavy lifting (OpenCV and Tensorflow, for example) and the Pico for the hardware control.

As an example, can we get a "blink LED" example that runs from a Python script on the Pi and blinks the Pico LED over serial USB connection?  Or slightly more advanced, how to program the Pi to send PWM servo position updates to the Pico.  I think expanding this kind of documentation would help a lot for the beginners trying to expand their understanding of how this stuff all works.

Thanks for all your hard work on this!  I'm really looking forward to receiving the Pico and starting to work with it!

lurch, 2021-01-31:
The MicroPython REPL on Raspberry Pi Pico is "just" a USB-serial connection (see "Chapter 2. Connecting to the
MicroPython REPL" of https://datasheets.raspberrypi.org/pico/raspberry-pi-pico-python-sdk.pdf ), so you should be able to connect to that from a Python script running on the Raspberry Pi 4 by using [pyserial](https://pyserial.readthedocs.io/en/latest/), and use that to send commands to the Pico's REPL prompt?
Although maybe a pre-canned example of doing that would be useful? :shrug:  @aallan could you move this to https://github.com/raspberrypi/pico-micropython-examples please?

FunkyGandalf, 2021-01-31:
Yes!  As a beginner, examples are life.  Using Pyserial to send commands to the Pico's REPL sounds awesome...just...confused...how :)  I find examples really help clarify just what code is actually necessary to do that.  

The tutorial at https://projects.raspberrypi.org/en/projects/getting-started-with-the-pico is a great explanation for us beginners, but only explains how to do these things on the Pico itself or from the REPL terminal.  Maybe this would be best addressed there by adding a new section or another tutorial explaining how to do these same things through Pyserial to the REPL?

closed 2021-05-25 by aallan