raspberrypi/pico-feedback #45: Include the explanation of SPIx_Rx and SPIx_Tx nomenclature in the Raspberry Pi Pico datasheet

kiwih, 2021-01-31:
In the `Get started with MicroPython on Raspberry Pi Pico' book there is this great explanation of the SPIx_Rx and SPIx_Tx nomenclature:

>SPI TERMINOLOGY SPI requires four connections: one that takes data from the master device to the slave device, another that takes data in the opposite direction, plus power and ground. Two data wires mean that data can travel in both directions at the same time. These are usually called Master Out Slave In (MOSI) and Master In Slave Out (MISO). However, you will come across them with different names. If you look at the Raspberry Pi Pico pinout (Appendix B), they're referred to as SPI TX (Transmit) and SPI RX (Receive). This is because Pico can be either a master or slave device, so whether these connections are MOSI or MISO depends on the current function of Pico.

It would be great if something like this could also make its way into the `Raspberry Pi Pico` datasheet (https://datasheets.raspberrypi.org/pico/pico-datasheet.pdf), and/or the `Raspberry Pi Pico Python SDK` (https://datasheets.raspberrypi.org/pico/sdk/pico_python_sdk.pdf), neither of which explain this nomenclature. 
The RP2040 datasheet does explain it, although not nearly as clearly or as concisely as the Get Started book does, so it could probably also benefit from the explanation as well.

lurch, 2021-03-31:
> SPI TERMINOLOGY SPI requires four connections: one that takes data from the master device to the slave device, another that takes data in the opposite direction, plus power and ground.

Yikes, I think that should have said "clock and chip select"? (or at the very least "clock and ground") ping @benevpi 
This is on page 121 of the current PDF.

aallan, 2021-04-21:
Fairly sure we did something about this one. @lurch can you remember?

lurch, 2021-04-22:
I think we just discussed it internally, and people agreed that using SPI_RX and SPI_TX (instead of MOSI/MISO) is common parlance when an SPI interface can be used in both Master and Slave modes, but I don't think anyone added anything to the databooks? :shrug:

icebarf, 2022-09-26:
Hi, Still waiting for thus to be fixed! Stumbled upon this when I could not understand what TX and RX were for.

powertomato, 2023-02-15:
So this means:
SPI_RX = MISO, SPI_TX = MOSI, when RPi Pico is a master
SPI_RX = MOSI, SPI_TX = MISO when RPi Pico is a slave device

Could someone confirm, please?