raspberrypi/pico-feedback #48: Installing picotool on Mac - symlink creation missing step

te-sveji-freddo, 2021-02-01:
Hi, I was installing picotool (Appendix B) following the guide "getting started with pico" and I found that on macOS you need to add the picotool binary symlink to the /usr/local/bin folder in order to use as a command, otherwise macOS will  give the error "command not found".

The command is:
`sudo ln -s /Users/your-nickname/picotool/build/picotool /usr/local/bin`

Regards,
FT

kilograham, 2021-02-01:
there is an issue for this in picotool repo already

closed 2021-02-01 by kilograham

te-sveji-freddo, 2021-02-01:
> there is an issue for this in picotool repo already

Sorry, I can't find it

aallan, 2021-02-01:
It's at raspberrypi/picotool#4.

te-sveji-freddo, 2021-02-01:
> It's at [raspberrypi/picotool#4](https://github.com/raspberrypi/picotool/issues/4).

Thank you!