raspberrypi/pico-feedback #52: Datasheet has multiple sections where bullet lists are not rendered properly

Muximize, 2021-02-01:
For example, section 4.8.5.5. _Interaction with Dormant / Sleep mode_ contains this text:

> To  wake  the  chip  from  dormant  mode:  *  the  RTC  must  be  configured  to  use  an  external  reference  clock  (supplied  by  aGPIO pin) * Set up the RTC to run on an external reference * If the processor is running off the PLL, change it to run fromXOSC/ROSC * Turn off the PLLs * Set up the RTC with the desired wake up time (one off, or recurring) * (optionally) powerdown  most  memories  *  Invoke  DORMANT  mode  (see  Section  2.16,  Section  2.17,  and  Section  2.11.5.2  for  moreinformation)

Which I suppose should be:

> To  wake  the  chip  from  dormant  mode:
>- the  RTC  must  be  configured  to  use  an  external  reference  clock  (supplied  by  aGPIO pin)
>- Set up the RTC to run on an external reference
>-  If the processor is running off the PLL, change it to run fromXOSC/ROSC 
>- Turn off the PLLs 
>- Set up the RTC with the desired wake up time (one off, or recurring) 
>- (optionally) powerdown  most  memories  
>- Invoke  DORMANT  mode  (see  Section  2.16,  Section  2.17,  and  Section  2.11.5.2  for  moreinformation)

Other places where this seems to happen:

- Table 84. M0+processor coreregister set summary (PC)
- Table 415. SIE_STATUS Register (bits 31 and 18)
- Table 443. UARTLCR_H Register (bit 7)

lurch, 2021-02-01:
Thanks. The first one I'd already spotted and fixed myself, and I've just fixed the second one (so these will be both be visible in the next version of the PDFs). The other two are trickier, due to the way those tables are created, so will need a bit more thought to fix.

aallan, 2021-05-25:
@lurch Did this get fixed, or is it backlog?

lurch, 2021-05-25:
There's currently no mechanism for rendering bullet-points (or any other kind of markup) that might be in a register-description - these come from VRB, so we have no idea whether an asterisk is intended to be used as a bullet-point, or whether it's intended to be a literal asterisk. (We could possibly add some kind of heuristic to say that if the asterisk is the first character on a line then it should be treated as a bullet-point, but this will obviously need careful checking in case it hits any false-positives.)

So yeah, still in backlog.

aallan, 2021-05-25:
That sounds like a huge amount of effort for a corner case. Can you raise an Jira on it — if we don't have one already?

closed 2021-05-25 by aallan