raspberrypi/pico-feedback #54: Raspberry Pi Pico C/C++ SDK, 3.3.9. Output pass through

fivdi, 2021-02-01:
In release 1.2 of the Raspberry Pi Pico C/C++ SDK documentation, in section 3.3.9. Output pass through, on page 52, the following text appears:

```
but could potentially be some_language.some_some_group if the the language generator supports
```

It should be:
```
but could potentially be some_language.some_some_group if the language generator supports
```

lurch, 2021-02-02:
LOL! I stripped out about half a dozen occurrences of "the the" from our docs a couple of months ago, but I guess this must have snuck in since then. :laughing:

lurch, 2021-02-02:
LOL again. The reason this one slipped through my fingers is because the two "the"s appeared either side of a line-break :laughing: 
Will be fixed in V1.3 of the PDFs (which have no ETA yet).

closed 2021-02-02 by lurch