raspberrypi/pico-feedback #56: "Getting Started" links don't work in Firefox 85

scruss, 2021-02-02:
From page https://www.raspberrypi.org/documentation/pico/getting-started click on "Getting Started with MicroPython"

Expected result: going to the MicroPython resources

Actual result: page reload.

The target anchor #getting-started-with-micropython doesn't seem to be present in the page. The link doesn't open if requested in a new tab.

Page works as well as can be expected in MS Edge.

lurch, 2021-02-02:
Do you have javascript disabled or something?

aallan, 2021-02-02:
I've passed this on to the Foundation Digital Team, and they'll investigate as soon as possible.

scruss, 2021-02-02:
> Do you have javascript disabled or something?

No. What might have been happening is that the new link content loaded, but the screen didn't move or indicate that the page content had changed.

DanielBrierton, 2021-02-04:
Hi @scruss 

Just to confirm, am I right in thinking it is working correctly for you, and you just didn't notice the content change?

scruss, 2021-02-04:
Yes, it is. On a couple of my machines, the tabs were right on the bottom of page, so the content reload appeared to do nothing

closed 2021-02-04 by scruss