raspberrypi/pico-feedback #57: C SDK output to USB drops last chaarcter(s) in minicom

Hermann-SW, 2021-02-03:
This is output in minicom, after disconnecting and connecting minicom several times.
The connected message should appear in next row, not directly after "done":
```
Port /dev/ttyACM0, 05:39:13

Press CTRL-A Z for help on special keys               
                                                      
tud_cdc_connected()                                   
donetud_cdc_connected()                               
donetud_cdc_connected()                               
done
```

Code:
```
#include <stdlib.h>
#include "pico/stdlib.h"
#include <tusb.h>

int main(void) {
  stdio_init_all();
  while (!tud_cdc_connected()) { sleep_ms(100);  }
  printf("tud_cdc_connected()\n");
  printf("done\n"); fflush(stdout);

  return 0;
}
```

Hermann-SW, 2021-02-03:
This is a bug report, not a question

closed 2021-05-25 by aallan

lurch, 2021-05-25:
@aallan Should this be re-opened and moved to pico-sdk ?

aallan, 2021-05-25:
> Should this be re-opened and moved to pico-sdk ?

I've seen no further reports of this happening, so I'm presuming it's user-specific issue rather than a general problem

lurch, 2021-05-25:
@Hermann-SW Was this just a "glitch", or is it still reproducible with the latest `develop` version of Raspberry Pi Pico SDK?