raspberrypi/pico-feedback #58: Possible error in Getting Started with Raspberry Pi Pico

osiiris02, 2021-02-03:
Under 8.2.3. Building "Hello World" from the Command Line it says

> Then set the path to the SDK as follows, 
`C:\Users\pico\Downloads> setx PICO_SDK_PATH "..\..\pico-sdk"`

This caused issues further along and I believe it should be ..\pico-sdk. I may have made another mistake before that but that's what worked for me.

lurch, 2021-02-03:
I've just had a quick look, and I believe the instructions as-written are correct.

If you follow the instructions in 8.2.2 correctly, you should have:
```
C:\Users\pico\Downloads\pico-sdk
C:\Users\pico\Downloads\pico-examples
```
(except obviously with your own username rather than `C:\Users\pico\`).
The `PICO_SDK_PATH` variable is used by the `cmake` command, so if you run the `cmake -G "NMake Makefiles" ..` command from inside the `C:\Users\pico\Downloads\pico-examples\build` folder as shown in 8.2.3, then the relative-path from there back to `pico-sdk` **is** `..\..\pico-sdk`. (A relative-path of `..\pico-sdk` from inside `C:\Users\pico\Downloads\pico-examples\build` would get you to `C:\Users\pico\Downloads\pico-examples\pico-sdk`, which isn't right **if** you've followed the rest of the instructions correctly).

Can you double-check which folders you have where?

michaelstoops, 2021-04-06:
I think there may be some issues that arise from the fact that the recommended PICO_SDK_PATH is relative. I move to recommend an absolute path.

The only benefit of a relative path, as far as I can tell, is having multiple SDK installations for the same user, and selecting different ones based on CWD. This could be necessary for those who are developing the SDK itself, but doesn't seem necessary for the intended consumers.

The cost of a relative path is that it only works with a certain folder structure. This clearly caused trouble for @osiiris02. I think it gave me some hassle as well, and I'd wager that it troubles a lot of others too.

aallan, 2021-04-06:
I don't think our intention was to recommend using a relative path for PICO_SDK_PATH. That's just the path to the `pico-sdk` checkout if you follow the instructions. Perhaps the assumption that folks would know to substitute the path to *their* checkout of `pico-sdk` was too much, I'll add an explicit note to the Getting Started guide somewhere around the first time we use the relative path.

closed 2021-04-06 by aallan

lurch, 2021-04-06:
Last week I was wondering if Windows supported the equivalent of `export ABSOLUTE_PATH=$(realpath $RELATIVE_PATH)` that you'd typically do in Linux, and it seems that it doesn't :slightly_frowning_face:  (there are various obscure path-modification-variable-suffixes, but AFAICT they only work from inside of a .BAT script)

n-mam, 2021-04-20:
I was trying to set up my pico-examples build on windows.. when I got this build error. Any pointers on how to move past this ?

[image]

lurch, 2021-04-20:
@n-mam Please create a new issue for your problem, rather than tacking onto the end of an already-closed issue. Thanks!

n-mam, 2021-04-20:
sure @lurch.. sorry about that. I see that arm also has a tool chain available for windows.. I'll try that before opening a new issue (I used the default arm visual studio toolchain above). thanks