raspberrypi/pico-feedback #59: Datasheet: 2.8.5 PICOBOOT

swetland, 2021-02-03:
The WRITE command, when used to write flash, is unreliable and hangs if the EXCLUSIVE_ACCESS command is not first used to gain exclusive access.

The WRITE command, when used to write flash, is unreliable if the EXIT_XIP command is not issued first.  No error occurs, but the written data is corrupted.

The FLASH_ERASE command's dAddr parameter ("address in flash") appears to mean 0x10000000... (address where flash will be visible via XIP), and not the offset from start of flash (0 based).  It would be helpful if this was made more explicit.

I suspect the READ command will not work from flash if ENTER_XIP is not issued first, but have not verified yet.

The WRITE and EXEC commands clearly are intended for code to to written to SRAM and executed.  However there is not (that I could find) any indication of which regions of SRAM are *safe* to use for this action (eg, how do I ensure I don't stomp on the ROM bootloader's data, bss, or stack regions?

kilograham, 2021-02-03:
> The WRITE command, when used to write flash, is unreliable and hangs if the EXCLUSIVE_ACCESS command is not first used to gain exclusive access.

You'll have to give a better repro, as this isn't obviously the case. If you are concurrently doing USB MSD writes then yes you should use EXCLUSIVE_ACCESS; it is what it is there for.

> The WRITE command, when used to write flash, is unreliable if the EXIT_XIP command is not issued first. No error occurs, but the written data is corrupted.
>
> The FLASH_ERASE command's dAddr parameter [...]
I> suspect the READ command will not work from flash if ENTER_XIP is not issued first, but have not verified yet.

We can perhaps improve the documentation here.

> The WRITE and EXEC commands clearly are intended for code to to written to SRAM and executed. However there is not (that I could find) any indication of which regions of SRAM are safe to use for this action (eg, how do I ensure I don't stomp on the ROM bootloader's data, bss, or stack regions?

the bootloader stuffs all of that away in a corner of the USB internal RAM, so you have full access to main RAM & XIP cache (if you aren't currently downloading a UF2 - which again you should use EXCLUSIVE_ACCESS)

swetland, 2021-02-03:
I'll investigate the first item in a bit more detail, it may be some other issue with my code I fixed simultaneously with starting to make use of the exclusivity request.  Definitely not using UMS (and would love a way to disable it entirely -- it seems to be making the Linux UMS driver very unhappy when the pico becomes non-responsive due to debugger attach, etc, and for the first time in years I had to reboot a Linux workstation due to the USB stack catching fire).

The intent here was primarily to request clarification / updates to the documentation.

swetland, 2021-02-04:
Cannot repro first item.  Striking it.

kilograham, 2021-02-04:
You can disable UMS if you programmatically reboot the USB into BOOTSEL mode.. second parameter to reset_usb_boot()

In case you missed it picotool uses PICOBOOT and can run a binary via `picotool load -x <binary>`

swetland, 2021-02-04:
I did miss that.  Nice.  

I glanced at picotool, but wanted something that didn't depend on cmake, a full sdk, etc, so ended up throwing together a cheesy little thing just to blast new codes onto the device: https://github.com/swetland/mdebug/blob/master/tools/picoboot.c

kilograham, 2021-02-04:
ha very nice... nice to see it being used; something i popped into the ROM long ago as it seemed like it would be useful.

swetland, 2021-02-04:
I was impressed with UF2 as a workaround for a lot of the gotchas with UMS emulation, but learned the hard way (again) why I've always avoided that and CDC ACM for MCU projects -- because desktop USB kernel drivers turn out to be shockingly fragile and blow up like crazy when devices fail to respond.  The support for userspace USB access (at least in Linux) is much more hardened against this and survives devices going off the deep end with much more grace.

closed 2021-05-25 by aallan