raspberrypi/pico-feedback #60: XIP cache already enabled in "RAM only" boots (preventing that memory region from being used)

majbthrd, 2021-02-03:
This seems like a possible undocumented errata issue...

Section 2.8.4.2 "UF2 Format Details" indicates how a "RAM only" binary can use the "Flash Cache" memory region (aka XIP_SRAM).  It says: "since flash is not being targeted, the Flash Cache is available for use as RAM with same properties as Main RAM"

In my testing, any "RAM only" image seems like it can only be located in "Main RAM"; any attempts to boot into "Flash Cache" fails.  The only explanation seems to be that the cache is being enabled prior to executing the "RAM only" image... either:

a)	due to boot ROM code execution or 
b)	due to it not being cleared during the watchdog reset used to boot into RAM only images

I am [providing diagnostic code](https://github.com/majbthrd/pico-cachedetector) that is the bare minimum (no pico-sdk) so that the SDK libraries couldn't possibly interfere.  It shows xip_ctrl->CTRL has bit 0 (EN) set when doing a RAM only boot.

kilograham, 2021-02-04:
Hmm; will have to take a look; i have to admit we never tried that (very much in the niche usage arena)... presumably you can make it work for now by putting a small trampoline in main RAM

majbthrd, 2021-02-04:
That's what I did as a workaround: the .uf2 loads into SRAM, the first few instructions disable the cache, then it copies itself into XIP_SRAM before branching there.  I fit a [RP2040 USB CMSIS-DAP debugger](https://github.com/majbthrd/pico-debug/)  (including stack) entirely in XIP_SRAM, leaving the 264kB SRAM completely available for the user app being debugged.  So, niche it may be, but hopefully it will useful to would-be RP2040 users.

kilograham, 2021-02-05:
note it should let you download data into XIP cache and RAM, no? so you should have to copy, but either way.

kilograham, 2021-02-05:
That is super cool (something similar has been somewhere on our list!) ... i'll take it for a spin when i'm back from vacation

lonetech, 2021-02-05:
That debugger is certainly cool, but I can't help wondering if it could be squeezed in to <4KiB? Because the USB block has 4KiB of which 3.5KiB are probably unused; and you're already using the USB peripheral. It's mapped at USBCTRL_DPRAM_BASE.

majbthrd, 2021-02-05:
@lonetech, it is certainly much easier said than done.  I have written a couple of 1kByte USB bootloaders, and back in 2013, I did a CMSIS-DAP implementation in 8kBytes for a $1 PIC USB 8-bit microcontroller.  I had to write the PIC CMSIS-DAP implementation from scratch, and it took A LOT of hand-optimization to get it *and* the USB stack into that 8kBytes.  If there were boot ROM code in the rp2040 that could be leveraged to implement the USB functionality, or it was a custom non-CMSIS-DAP debug protocol using just USB EP0, perhaps there might be a shot at getting it to fit.  However, the odds are against it.

Let's see if anyone even cares about pico-debug before going down that lengthy and arduous road (likely with a dead-end).  Seven years ago, I thought there would be strong interest in a $1 BOM addition to hobbyist ARM Cortex-M MCU boards, and boy was I seriously wrong.  I did pico-debug because I read the RP2040 datasheet and immediately thought it was compelling to have a software-only CMSIS-DAP debugger without any additional hardware; we'll see in time if this was another fool's errand of mine.

At the risk of stating the obvious, it is worth pointing out that the present pico-debug can be put anywhere* there is 16kBytes free.  Since the rp2040 has a cavernous 264kBytes, one could either:
a)	carve out 16kBytes from the 264kByte SRAM and leave flash cache enabled, or
b)	leave all 264kB free, use the 16kBytes flash cache, and disable flash cache

I picked (b) initially because I have no control over pico-sdk, and it currently spreads variables all over the entire 264kByte SRAM address space.

*elf2uf2 seems artificially picky about where images are placed, so it is perhaps not quite anywhere

daveythacher, 2021-02-12:
Switching automatically may not be supported. Documentation confusion.

Question for internal is:
Does the hardware actually support automatically loading into cache then marking the cache as SRAM upon completion of UF2 boot?

daveythacher, 2021-02-12:
You may still need flash even with RAM execution. The boot loader could detect this from UF2 format. Have you looked into that?

majbthrd, 2021-02-13:
@daveythacher, flash is most certainly NOT needed with RAM execution.  I have a working and useful [pico debugging tool](https://github.com/majbthrd/pico-debug) that operates purely in RAM.  It will even run in XIP, but only when the .uf2 loads into RAM and then disables cache copies itself into XIP.

lonetech, 2021-02-13:
I had a little browse around in the boot rom sources https://github.com/raspberrypi/pico-bootrom and it appears the bootloader does allow to load into the XIP SRAM, but doesn't know how to disable the XIP cache to make that valid. So probably elf2uf2 needs an extra option to load a function into main RAM that clears bit 0 (EN) in XIP_BASE[CTRL], and call it using the PC_EXEC command, before writing into the XIP SRAM. PC_EXEC does a Thumb subroutine call, AFAICT; it doesn't have a great amount of stack but can return to the bootloader.

lonetech, 2021-02-13:
I think I finally found the issue. The UF2 bootloader uses safe_reboot to start the loaded program, using a watchdog reset. That reset will enable the XIP cache, invalidating all data stored in XIP SRAM. So while UF2 allows writing into XIP SRAM, it'll be inaccessible when it starts up again. The code path in boot ROM that disables XIP cache only runs when it goes into USB bootloader mode. The boot ROM could be extended to detect if XIP SRAM was written, and use a watchdog scratch register to instruct the boot ROM post reset to disable XIP cache before doing the jump to the program start. 

This would require new code in check_wdog and somewhere before _write_uf2_page_complete. Obviously the existing mask ROMs can't be updated thus. 

However, a workaround is possible: Have a small startup trampoline in main RAM, which clears the XIP cache enable bit, then jumps into the XIP SRAM. This would be possible to write into a UF2, requiring only a custom linker script and startup code. The reset shouldn't wipe the XIP SRAM contents, according to the RP2040 datasheet XIP: FLUSH Register documentation.

majbthrd, 2021-02-17:
Testing @lonetech 's code interpretation required [a patch to elf2uf2 in pico-sdk](https://github.com/raspberrypi/pico-sdk/pull/140).

I can confirm that, with that modification, it is possible to have .uf2 blocks that load directly into XIP_SRAM intact providing there is also a MAIN_RAM trampoline that immediately switches off the cache.

However, in the process of testing I discovered something odd happening with the disabling the cache:

Unless *after* the write of 0x0 to XIP_CTRL to disable the cache there are many CPU cycles (around a hundred) *OR* there is a read of XIP_FLUSH (latter is my current preference), writes to that region can cause corruption of the cache memory.  The datasheet indicates there will be bus stall when reading XIP_FLUSH after a flush request, but there must be some other event going on after a cache disable that is helped by a XIP_FLUSH read.

Wren6991, 2021-02-17:
> Unless after the write of 0x0 to XIP_CTRL to disable the cache there are many CPU cycles (around a hundred) OR there is a read of XIP_FLUSH (latter is my current preference), writes to that region can cause corruption of the cache memory.

This does sound like you are accessing the cache-as-SRAM alias whilst the cache flush is still in progress (takes just over 2048 cycles). Cache flush is triggered by `XIP_FLUSH` and by the XIP block coming out of reset, so you would be able to catch the tail end of the flush on a watchdog boot.

>  The datasheet indicates there will be bus stall when reading XIP_FLUSH after a flush request,

Yes, accesses to the cached XIP window will generate a stall if they take place during a flush, which makes it easier to flush the cache whilst concurrently executing through it. The cache-as-SRAM alias doesn't have this interlock.

> but there must be some other event going on after a cache disable that is helped by a XIP_FLUSH read.

Yes the cycling of the tag memory during flush will stop cache-as-SRAM mode from selecting the correct way of the cache set. A dummy `XIP_FLUSH` read is the correct way to fence off on this, but this all needs to be documented.

Wren6991, 2021-02-17:
Added an errata entry for this:

[image]

And a caution under 2.6.3.2 for cache flushing and CSRAM:

[image]

Will appear in next docs release.

closed 2021-05-25 by aallan