raspberrypi/pico-feedback #61: RP2040 PI PICO - DMA

irishconger, 2021-02-04:
Cant really see how this works for RP2040 PI PICO.

From other controllers, you would have:
SOURCE BUFFER LOCATION -- with or without auto-increment
DESTINATION BUFFER LOCATION -- with or without auto-increment
NUMBER OF TRANSFERS (e.g. transfer 1k values)
TRIGGER EVENT for DMA transfer (e.g. ADConversion is done)
REPORT EVENT ON NUMBER OF TRANSFERS COMPLETE (e.g. generate interrupt)
AUTO-TRIGGER ON NUMBER OF TRANSFERS COMPLETE (e.g. swap to alternate buffer -  this would allow for seamless 'continuous' transfer of ADC samples)

How do these map to RP2040 PI PICO?

g

lurch, 2021-02-04:
Are there _specific_ questions you have that section "2.5. DMA" of https://datasheets.raspberrypi.org/rp2040/rp2040-datasheet.pdf doesn't answer? Have a look also at https://github.com/raspberrypi/pico-examples/tree/master/dma

irishconger, 2021-02-04:
Thanks Andy.... I was using out of date documentation.
Indeed, most of the functionality I was looking for is covered in the doc u
referenced.
g

closed 2021-02-04 by aallan