raspberrypi/pico-feedback #63: Document how to adjust system clock without decreasing peripheral clock

Hexxeh, 2021-02-04:
For users who want to push the boundaries (and void their warranties!) it'd be great if you could document how to link clk_peri to a divided PLL, so that users who do this can stop the peripheral clock dropping to 48Mhz and losing most of the gains they made with a higher system clock elsewhere (eg waiting on SPI).

Wren6991, 2021-02-04:
Note there is no separate divider on `clk_peri` (it is usually a subsidiary of `clk_sys`) so you will need to, uh, overclock your UART as well.

Hexxeh, 2021-02-04:
Thanks! I'm using the USB CDC UART, which seems to have a separate clock, so that would be unaffected I hope?

I can see from the example how you can reduce the clk_sys whilst keeping clk_peri at the default, but is it possible have clk_sys higher than clk_peri if clk_peri has no divider on it?

Hexxeh, 2021-02-04:
I just noticed in the clocks section of the datasheet on page 200, there's a clocks diagram that shows a divider between clk_peri and sys_pll, but the comment suggests no divider is possible as I commented. Am I misunderstanding the comment or is the datasheet wrong? Probably the former :)

liamfraser, 2021-02-08:
The best place to look for now is in the clocks registers. clk_peri has no dividers, so your choices are:

clk_sys (probably configured to pll_sys)
clksrc_pll_sys (whatever you are overclocking to)
clksrc_pll_usb (48MHz)
rosc (~6.5MHz but not a fixed frequency so not ideal for serial comms)
xosc (12MHz)

The USB CDC UART is a software uart using the USB controller so it doesn't have a peripheral clock other than the 48MHz USB clock for the USB controller.

liamfraser, 2021-02-08:
The other option (if you switch your USB UART out for a hardware UART) is to run your USB PLL at a faster speed. In theory you should be able to run your USB PLL at 133MHz and run clk_peri from that.

closed 2021-05-25 by aallan