raspberrypi/pico-feedback #64: Getting Started With Raspberry Pi - Chapter 7 - Create Your Own Project - CMake missing enable_language

evinrude, 2021-02-05:
Documentation says to create a CMakeLists.txt with the contents

```
cmake_minimum_required(VERSION 3.13)

include(pico_sdk_import.cmake)

pico_sdk_init()

add_executable(test test.c)

pico_enable_stdio_usb(test 1)

pico_enable_stdio_uart(test 1)

pico_add_extra_outputs(test)

target_link_libraries(test pico_stdlib)
```

Without adding ```enable_language( C CXX ASM )```  the cmake command will yield the same error many times.

CMake Error: Error required internal CMake variable not set, cmake may not be built correctly.
Missing variable is:
CMAKE_ASM_COMPILE_OBJECT
CMake Error: Error required internal CMake variable not set, cmake may not be built correctly.
Missing variable is:
CMAKE_ASM_COMPILE_OBJECT
CMake Error: Error required internal CMake variable not set, cmake may not be built correctly.
Missing variable is:
CMAKE_ASM_COMPILE_OBJECT
CMake Error: Error required internal CMake variable not set, cmake may not be built correctly.

aallan, 2021-02-05:
Hey, looks like the example in the documentation is missing a `project( )` line, so the CMakeLists.txt file should read,

```
cmake_minimum_required(VERSION 3.12)

include(pico_sdk_import.cmake)

project(test_project)

pico_sdk_init()

add_executable(test
    test.c
)

pico_enable_stdio_usb(test 1)
pico_enable_stdio_uart(test 1)

pico_add_extra_outputs(test)

target_link_libraries(test pico_stdlib)
```

I'll make the fix and it'll be in the next PDF release. Thanks!

aallan, 2021-02-05:
Hey. I'm not sure where the above text comes from? The `CMakeLists.txt` file for the "Creating your own Project" chapter in the documentation looks like this,

```
cmake_minimum_required(VERSION 3.12)

include(pico_sdk_import.cmake)

project(test_project)

pico_sdk_init()

add_executable(test
    test.c
)

pico_enable_stdio_usb(test 1)
pico_enable_stdio_uart(test 1)

pico_add_extra_outputs(test)

target_link_libraries(test pico_stdlib)
```

see the bottom of p.27 and the top of p.28 in the latest V1.2 documentation release. This compiles cleanly on my Mac where CMake is a bit more lenient. However it should properly be,

```
cmake_minimum_required(VERSION 3.13)

include(pico_sdk_import.cmake)

project(test_project C CXX ASM)
set(CMAKE_C_STANDARD 11)
set(CMAKE_CXX_STANDARD 17)
pico_sdk_init()

add_executable(test
    test.c
)

pico_enable_stdio_usb(test 1)
pico_enable_stdio_uart(test 1)

pico_add_extra_outputs(test)

target_link_libraries(test pico_stdlib)
```

I'll go ahead and update the instructions in the text and this fix will be in the next docs release (V1.3). 

Thanks!

closed 2021-02-05 by aallan