raspberrypi/pico-feedback #65: Q: Synchronization of PIO state machines

henrygab, 2021-02-05:
<details><summary>Background of question</summary><P>

The datasheet documents that, on a per-state-machine basis, the clock rate used for the state machine can be set using a divisor of the system clock.  The division has both a 16-bit integer (`n`) and 8-bit fractional (`f`) portion, all of which is put into `((pio_sm_config *)c)->clkdiv`.  So far, so good.

The datasheet also indicates that, when the 8-bit fractional portion is non-zero, then the state machine will run every `n` or `n+1` cpu clock cycles (depending on the fractional part; accumulator?).  By itself, if running at 133MHz, this is less than 8ns difference, which should be OK.

Sometimes, it desired to synchronize multiple state machines.  A documented method includes the state machine instructions waiting on an interrupt.  Then, since each instruction is a single (state-machine) cycle, it would initially seem that identical state machine executions (and state) would continue to run in perfect lock-step.  Great!

</details><hr>

<details><summary>Simple example PIO assembly</summary><P>

The following should toggles four GPIO outputs at a fixed rate.
The first simply starts execution as soon as the state machine is configured.
The second waits for IRQ 7 to first, to attempt to synchronize across state machines.

```asm
.program no_synchronized_start
.side_set 4

wrap_target:
    nop side 0xF;
    nop side 0x0;
.wrap

.program synchronized_start
.side_set 4

    wait 1 irq 7 side 0x0; wait for IRQ7 to fire
wrap_target:
    nop side 0xF;
    nop side 0x0;
.wrap
```
</details><hr>

The question is, if each state machine is initialized with the same clock divisor (integer and fractional part), in which conditions will the state machine cycles be, and remain, 00% synchronized (CPU clock accurate ... not even deviating by a single CPU cycle)?

<details><summary>Definition of the three conditions</summary><P>

I can think of three conditions that could influence this:
1. Did the state machines **_synchronize execution_** via the `wait` instruction? _(maybe still yes, if the fractional divisor causes `+1` based on an shared counter across each (or both) PIO banks)_
2. Are the state machines in the **_same PIO bank_**? _(if not, maybe there `+1` changessuch as where based on a counter distinct to each PIO bank)_
3. Is the clock divisor's **_fractional part zero_**? _(may affect answer when two state machines are **_NOT_** in same PIO bank)_

</details><hr>

**Q: Which combination of conditions could result in 1-cpu cycle of jitter between state machine execution cycles?**

Condition | Case7 | Case6 | Case5 | Case4 | Case3 | Case2 | Case1 | Case0
-- | -- | -- | -- | -- | -- | -- | -- | --
State   machines start synchronized | TRUE | TRUE | TRUE | TRUE | FALSE | FALSE | FALSE | FALSE
State machines in same   PIO bank | TRUE | TRUE | FALSE | FALSE | TRUE | TRUE | FALSE | FALSE
Clock_div   fractional part is zero | TRUE | FALSE | TRUE | FALSE | TRUE | FALSE | TRUE | FALSE
**fully synchronized?** |   |   |   |   |   |   |   |

Wren6991, 2021-02-05:
Things to note here:

- Each clock divider is a free-running piece of hardware -- once you set it running it will generate a certain pattern of clock enable pulses, completely determined by its initial state, regardless of what instructions the SM executes
- To have an identically-phased "clock" on all SMs (including the same fractional jitter), you must start all of their clock dividers from the same initial state at the exact same time
- Disabling a state machine (clearing its bit in CTRL_SM_ENABLE) holds its clock divider in synchronous reset, and setting the enable bit releases the clock divider to begin running from its initial state

So *as long as your SMs are in the same PIO block* you can get their execution "clocks" exactly synchronised by starting all SMs with a single CTRL_SM_ENABLE write. If their execution clocks are synchronised, you can then get the program execution in precise lockstep using e.g. a `wait irq`.

Note also that the CTRL_CLKDIV_RESTART strobe allows you to resync clock dividers following a change to the clock divisor whilst the state machines are running.

There is no mechanism to do this across state machines in different PIO blocks.

henrygab, 2021-02-06:
@Wren6991, thank you for the excellent and straightforward explanation!

if you have time for one more clarifying question, then given:

1. All four state machines in one PIO block have identical clock divisor setting
2. All four state machines were previously **_un_**synchronized
3. All four state machines are executing the same wait (e.g., on an IRQ)

Would using the `CTRL_CLKDIV_RESTART`, followed by setting the IRQ, result in identically-phased "clocks" for all four state machines?

Wren6991, 2021-02-08:
Yes, the clock dividers would all start from 0 internal state at the same instant, leaving all four SMs in the `wait irq` state with their execution clocks synchronised. They will then all see and clear the IRQ simultaneously, then drop into their next instructions simultaneously.

Wren6991, 2021-02-08:
I'll add some info to the datasheet based on this documentation, will post it here when done.

Wren6991, 2021-02-12:
There is some additional info in the register listings now e.g.: 
[image]

And this will appear in the next datasheet release. There is also a lot of extra information in the pio.h doxygen now e.g. here:

https://github.com/raspberrypi/pico-sdk/blob/a2d82431e6e34d938be3085fd9f3d53ef5dafb2b/src/rp2_common/hardware_pio/include/hardware/pio.h#L563

Would like to add a proper datasheet section at some point, with diagrams, but I don't think it will happen soon. Hope this clears up the original questions.

closed 2021-02-12 by Wren6991

Corallus-Caninus, 2021-06-19:
https://raspberrypi.github.io/pico-sdk-doxygen/group__hardware__pio.html#gac3797a9f7e80606152eb44af9ca96398

henrygab, 2022-09-29:
> This seems to do exactly that in 1 line and 1 clock but has a distinct parameter for PIO.

[`pio_enable_sm_mask_in_sync()`](https://github.com/raspberrypi/pico-sdk/blob/426e46126b5a1efaea4544cdb71ab81b61983034/src/rp2_common/hardware_pio/include/hardware/pio.h#L663-L679)

Yes, it addresses my [three-condition question](https://github.com/raspberrypi/pico-feedback/issues?q=is%3Aissue+PIO+is%3Aclosed#issuecomment-774377084).

I wonder ... it seems to suggest an even more edge-case scenario is supported:

1. First SM was executing, and could currently be stalled... or it could also be running (so long as it is executing instructions where massive clock jitter for one instruction is OK).
2. CPU can disable / load / configure / replace the second SM ... totally replacing what it does.
3. CPU calls this function with the PIO, and a `mask` indicating both the first and second SMs need to be synchronized.

When that writes the register, the first SM will have one instruction with potential clock jitter (but most likely it's stalled in a safe place).  Meanwhile the second SM starts execution immediately ... but synchronized with the first SM. 
 Importantly, the first SM will not lose any execution state ... continuing in sync with the second SM.

Nice!  I can envision this being used to dynamically swap between two SM instruction sets for the second SM...

tybasuleman, 2024-02-01:
Can anyone help me out. I am trying to sync my 2 state machines with a clock signal so when clock is ON, State Machine one sends a signal and when it´s OFF, the other one sends a signal. Signal here is PDM bits. The problem I think I understand is that every instruction is carried out for one complete clock cycle, not half of it as per my requirement and I cannot say out pins, 0.5 or something per say. It has to be between 1 and 32. 
So I am unable to turn the state machine off mid way. We cannot use two wait instructions in one pio file as well otherwise that would have done the job. If I am approaching the problem wrong then please correct me.
Thanks

lurch, 2024-02-01:
@tybasuleman This issue has been closed for well over a year, so you should probably create a new issue?

tybasuleman, 2024-02-01:
> @tybasuleman This issue has been closed for well over a year, so you should probably create a new issue?

Alright Thanks