raspberrypi/pico-feedback #66: RP2040 datasheet - PIO documentation issue

caleis, 2021-02-05:
RP2040 datasheet says the followings:
3.4.2.2. Operation
Set program counter to Address if Condition is true, otherwise no operation.
**Delay cycles on a JMP always take effect, whether Condition is true or false**, and they take place after Condition is evaluated
and the program counter is updated.

According to my experience with PIO programming done on RPI Pico, when the jump takes place the delay value is not taken into account. observed it in conjunction with jump when X does not equal to zero, post auto-decrement (e.g. jmp x-- [5])

Could you please check the validity of the documentation and please correct or amend.
Thanks!

Wren6991, 2021-02-05:
The datasheet is correct as far as I know. Could you please provide a complete program that demonstrates the behaviour you're seeing?

Here is my repro, in MicroPython:

```
from rp2 import PIO, StateMachine, asm_pio
from machine import Pin

@asm_pio(sideset_init=(PIO.OUT_LOW,PIO.OUT_LOW))
def jmp_delay():
    set(x, 3)       .side(2) [7]
    label("l")
    nop()           .side(0) [1]
    jmp(x_dec, "l") .side(1) [3]
    

sm = rp2.StateMachine(0, jmp_delay, freq=1_000_000, sideset_base=Pin(16))
sm.active(1)
```
As you can see the bit 0 pulse is 4 microseconds wide on all 4 loop iterations:

[image]

caleis, 2021-02-06:
Hi Luke,

appreciate the quick repro and feedback. I should have made a similar simple test myself. No I verified in my code as well that the behaviour is as documented. So my apologies for wasting your time.
I made a double mistake and incorrectly concluded that the behaviour is different than documented. Please close the case.
Please find the code below:

```
.program recs80
.side_set 1                      ;side set configuration (always/opt)
; An infrared transmit program with RECS-80 coding
recs80start:
   pull              side 0      ; Load next 32-bit word and set output to 0, stall if fifo empty
   irq      set 0    side 0      ; trigger aux state machine
   set      y, 10    side 0      ; outer loop counter initialisation, 11 bits altogether
nextbit:
   set      x, 5     side 0      ; Preload bit counter
pulses:                          ; This loop will run 6 times
   nop               side 1      ; high pulse
   jmp x-- pulses    side 0 [1]  ; low pulse and jump
   set      x, 21    side 0      ; 576 units - pulse train 18 base timing units + 4 from instructions
fixdly:
   nop               side 0 [12] ; inside loop is 13+12 = 25 clocks
   jmp x-- fixdly    side 0 [11] 
   out      x, 1     side 0 [1]  ; Shift 1 bit from OSR to x
   jmp !x, bitdone   side 0
   set      x, 27    side 0 [7]  ; for pulse hi, 288 clocks needs to be added: 8 here + 280 in cycle
hiaddon:           
   jmp x-- hiaddon   side 0 [9]
bitdone:
   jmp y-- nextbit   side 0
   set      x, 5     side 0      ; Preload bit counter
lpulses:                         ; This is the closing pulse train
   nop               side 1      ; high pulse
   jmp x-- lpulses   side 0 [1]  ; low pulse and jump
   wait    1 irq 1   side 0	 ; wait till the end of entire IR code timing

.program recs80aux
.side_set 1                      ; side set configuration (always/opt)
; For more accurate timing, running on the same clock as main SM

r80auxstart:
   wait     1 irq 0  side 0	 ; main SM triggers when start transmitting a new code
   set      x, 15    side 1 [14] ; initialise outer loop. Total delay 16x16x54 clocks
raouter:
   set      y, 15    side 1 [13] ; initialise inner loop
rainner:
   nop               side 1 [15]
   nop               side 1 [15]
   nop               side 1 [15]
   jmp y-- rainner   side 1 [4]
   jmp x-- raouter   side 1
   irq      set 1    side 0	 ; when done signal main SM via IRQ 1
```

The issue was with the auxiliary (second) state machine, which needs to provide 121.65 msec delay, starting from the first pulse emitted by the first SM. 
I calculated the timing wrong AND then erroneously replaced the 
  ` jmp y-- rainner   side 1 [4]`
with 
`   nop side 1 [4]
    jmp y-- rainner side 1`

Then falsely concluded that jmp delays are not as documented (as the jmp takes up one clock, in the replacement code the nop should have had delay [3] instead of [4]...

[image]

closed 2021-02-06 by lurch