raspberrypi/pico-feedback #67: Documentation: PIO: JMP/WAIT: GPIO mapping of PIN source

soundpaint, 2021-02-06:
Hi,
referring to the RP2040 datasheet,

Sect. 3.4.2.2 (JMP) says: "JMP PIN branches on the GPIO selected by EXECCTRL_JMP_PIN ."
Sect. 3.4.3.2 (WAIT) says: "PIN : Input pin selected by Index . This state machine’s input IO mapping is applied first"
Sect. 3.5.6, Fig. 48 depicts an "IN base" control value "for WAIT, JMP" between the two mapping architectural blocks.

(Also note, that above the "IN base" words there is a small, irritating vertical line, indicating that "IN base" were flowing from the upper architectural block to the lower one.  I guess, this vertical line is wrong, since "IN base" is just a parameter to the lower block as e.g. "OUT base" is a parameter to the upper block, right?)

So, is I/O mapping for the WAIT PIN also mapped via the EXECCTRL_JMP_PIN?  Or is there an extra EXECCTRL_WAIT_PIN for the WAIT PIN?  Or are EXECCTRL_JMP_PIN and EXECCTRL_WAIT_PIN (if there is anything like that) just synonyms for the "IN base" value?

Best regards & thank you,
Jürgen

lurch, 2021-02-06:
`IN_BASE` seems to be set via e.g. `SM0_PINCTRL.IN_BASE` (Table 397) and `JMP_PIN` seems to be set via e.g. `SM0_EXECCTRL.JMP_PIN` (Table 393) which additionally clarifies "The GPIO number to use as condition for JMP PIN. Unaffected by input mapping.".
I wonder if this _maybe_ implies that the lower block in Figure 48 should have a `JMP_PIN` input-parameter in addition to an `IN_BASE` input-parameter?

In 3.4.3.2 it looks like for the `WAIT` keyword you can either select "absolute GPIO mapping" `00: GPIO: System GPIO input selected by Index. This is an absolute GPIO index, and is not affected by the state machine’s input IO mapping.` or "offset GPIO mapping" `01: PIN: Input pin selected by Index. This state machine’s input IO mapping is applied first, and then Index selects which of the mapped bits to wait on` (where I assume "input IO mapping" refers to `IN_BASE`?). By contrast, the `JMP` keyword only has "absolute GPIO mapping".

Apologies in advance to @Wren6991 if I've got this totally wrong! :wink:

soundpaint, 2021-02-07:
- +1 for the idea to add "JMP_PIN" as input parameter in Fig. 48.
- Yes, JMP_PIN obviously refers to EXECCTRL JMP PIN, and IN_BASE to EXECCTRL IN_BASE.  Sorry about that! :-)
- My core question still remains open: For the WAIT instruction, how does the IO mapping apply?

cu, Jürgen

Wren6991, 2021-02-08:
> there is a small, irritating vertical line, indicating that "IN base" were flowing from the upper architectural block to the lower one.

Yes that shouldn't be there, looks like it was added during redraw. Thanks

> +1 for the idea to add "JMP_PIN" as input parameter in Fig. 48.

Yes that should be added

> For the WAIT PIN instruction, how does the IO mapping apply?

For WAIT PIN, you are indexing the input bus used by the IN PINS instruction. In other words, the pin you are waiting on is selected by adding the WAIT's `Index` argument to `IN_BASE`, modulo 32.  On the other hand, the WAIT GPIO instruction just picks a GPIO number using the `Index` argument (no `IN_BASE` addition).

I'll get that diagram fixed up and add some clarifying text to the WAIT PIN description.

Wren6991, 2021-02-08:
New wording in 3.4.2.2:
[image]

New wording in 3.4.3.2:
[image]

Wren6991, 2021-02-08:
[image]
This is how fig. 48 should look

aallan, 2021-02-08:
@Wren6991 Should I send the above figure to Jack?

Wren6991, 2021-02-08:
I'm just compiling an email now, there are some stray waveform diagrams
that aren't redrawn.

Wren6991, 2021-02-08:
This is with Jack now for redrawing. Text changes are in the upstream datasheet. Seeing a lots of thumbs up from @soundpaint  so I'm guessing this is sorted :)

closed 2021-02-08 by Wren6991