raspberrypi/pico-feedback #68: Documentation: Deprecated terminology in the I2C SDK documentation

Pinkyatacme, 2021-02-06:
It's only a minor issue to some, but the "master/slave" terminology was formally deprecated over a year ago. The I2C documentation should really be updated to use the correct "main/secondary" terms.

ebenupton, 2021-02-08:
Could you provide a reference for this please?

Pinkyatacme, 2021-02-08:
On further checking, my comments around the formal deprecation are not actually correct - I was conflating this with another protocol definition that went through this process. The use of this terminology has been, for some time now, under review by the IEEE, and it may be appropriate to start updating the documentation (and possibly code declarations) as "the right thing to do".

ebenupton, 2021-02-08:
Thanks for the clarification. We will keep an eye on this: at the point where master and slave are no longer in general use to describe bus roles, we will of course need to update.

closed 2021-02-09 by aallan