raspberrypi/pico-feedback #69: Documentation: PIO: PINCTRL OUT_COUNT

soundpaint, 2021-02-07:
RP2040 datasheet, Sect. 3.7, Table 397 says:
"25:20 OUT_COUNT The number of pins asserted by an OUT. Value of 0 → 32 pins"

Really?  Having 6 bits (25:20) implies potentially having values 0..63.  I would have expected 5 bits only.

soundpaint, 2021-02-07:
By the way, in case it should read "24:20", maybe functions sm_config_set_out_pins() and pio_sm_set_out_pins() in pio.h should clear the uppermost 6th bit when value 32 is passed in?

Wren6991, 2021-02-08:
32 in binary is `0b100000`, which is 6 bits wide. The datasheet is correct.

Setting to 0 means that state machine's OUT is disabled. Setting to 32 means all of the (maximum of) 32 pins writable by a state machine will be affected.

soundpaint, 2021-02-08:
The datasheet says "Value of 0 → 32 pins".  That is, instead of 0b100000, the value 0b0 should be written?  Then, it is still 5 bits.  

Also, what happens when writing values 33..63?  Is this equivalent to writing values 1..31?  Or are the values 33..63 reserved?

soundpaint, 2021-02-08:
Ah, ok, I got it know!  Sorry for the confusion! :-)
Still, the "Value of 0 → 32 pins" maybe could be a little bit more clarified (i.e. the difference between values 0 and 32)?

Wren6991, 2021-02-08:
Ohhh I see how it can be interpreted that way, as there are some other fields that are 5 bits, and map 0 to 32. In this case it refers to an inclusive range of 0 to 32. I'll clear that up. Sorry, I was hasty to close this.

Wren6991, 2021-02-17:
This is fixed upstream, and will appear in next docs release. Also fixed in the SDK headers on `develop`, which contain the same text as the next datasheet release.

closed 2021-02-17 by Wren6991