raspberrypi/pico-feedback #70: Inconsistent use of pullup and pull-up

PeterOGB, 2021-02-07:
In the RP2040 data sheet the hyphenated and non-hyphenated versions of "pullup/pull-up" are used.
The same applies for "pulldown/pull-down"

This means you have to search the document twice to find all the information !

aallan, 2021-05-25:
@lurch Did you get round to do a find-and-replace for this one?

lurch, 2021-05-25:
Not yet, it never floated to the top of my priority list!

JamesH65, 2021-07-02:
Just reported on the forums as well. https://www.raspberrypi.org/forums/viewtopic.php?f=144&t=315020

PeterOGB, 2021-07-02:
How is telling the poster on the forums it's been reported, but not telling them nothing has been done about it in five months helpful ?
Clearly no one thinks it's important so I'll just close it as "Wont Fix"

aallan, 2021-07-02:
It's on the list.

PeterOGB, 2021-07-02:
You are taking the piss !

dp111, 2021-11-19:
There is a third variant as well "pull up" / "pull down"

issa-tseng, 2022-06-11:
i tried to give this a go but either i'm stupid or the pdf documentations aren't public for contributions?

aallan, 2022-06-11:
> i tried to give this a go but either i'm stupid or the pdf documentations aren't public for contributions?

The source of the PDF documentation is not public.

issa-tseng, 2022-06-11:
okay, thanks for the clarification! was just hoping to give back a bit. :)

aallan, 2024-01-18:
Fixed in the next docs release.

closed 2024-01-18 by aallan