raspberrypi/pico-feedback #72: Pico PCB Feature Request: standard SWD debug connector (footprint)

swetland, 2021-02-08:
There's a reasonable amount of open space on the top side of the Pico PCB.

It would be really nice to at least drop a footprint for a standard 1.27mm pitch 2x5 SWD connector so folks using standard tooling could solder a header down and be good to go.

With GND/Vref/SWDIO/SWDCLK/nRESET you've got the ability to power (or voltage-match), debug, and hard-reset the board.

A UART0.TX capable pin routed to the SWO pin (maybe via a cut/resolder link) would be nice as a way to grab serial debug via the same connector.

swetland, 2021-02-14:
Additional notes, per discussion on Discord:
- GND/Vref/SWDIO/SWDCLK is the min viable connectivity needed for this to be useful -- Vref is important to ensure that debug interfaces with level shifters can adjust them properly or debug interfaces providing power can provide power.
- Since these four pins happen to be adjacent, it is possible to use a 2x2 1.27mm pitch header, provided there's adequate clearance for the overhang from the remaining 6 pins on the cable side.

Nice to have functionality in order from most to least useful
- SWO / UART0.TX -- debug output chatter on the same cable is always handy
- UART0.RX -- debug input too is a bonus
- nRST - I'm used to this not being present as most Cortex-M SWD setups allow for reliable system reset via SWD, and I believe this is the case with rp2040 as well

closed 2021-05-25 by aallan

Frank-Buss, 2022-10-09:
Is this marked as completed because the footprint for the debug connector was added? At least for the boards I bought recently it wasn't there.
A standard ARM Cortex 10 pin debug 0.05" micro header would be nice:
https://documentation-service.arm.com/static/5fce6c49e167456a35b36af1
Could be SMD. Doesn't increase the price, because not populated by default, but makes it easier to debug from a JLink debugger.

aallan, 2022-10-09:
> Is this marked as completed because the footprint for the debug connector was added? 

It was closed as inactive. We are very unlikely to add this header to the Pico. However if you're looking for RP2040-based boards with this, there are a large number of third-party vendors now shipping hardware with the chip, it's quite possible one of these vendors have what you're looking for.

JamesH65, 2022-10-09:
The Pico H, as shown here, has a pre-soldered SWD header. https://www.adafruit.com/product/5525

hattesen, 2022-12-28:
> The Pico H, as shown here, has a pre-soldered SWD header. https://www.adafruit.com/product/5525

The Pico H doesn't seem to have a SWD debug port footprint, but the [Adafruit Feather](https://www.adafruit.com/product/4884) does have the footprint ready to solder a 0.05" SMD header onto.

[image]

And the [SWD pin header](https://www.adafruit.com/product/752) is available from Adafruit.