raspberrypi/pico-feedback #73: pwm_set_gpio_level() clarification

blippy, 2021-02-08:
The C/C++ SDK documents the pwm_set_gpio_level(), but I think some clarification is required. 

RP2020 Datasheet section 4.5 (PWM: Programmer's Model) states that each slice has two copies of CC and TOP in order to achieve double-duffering. 

I think this means that pwm_set_gpio_level() guarantees that the level will be changed on the next wrap, not immediately. Can you confirm this? Also, can you please annotate the SDK accordingly so that its behaviour is clear to the reader?

Wren6991, 2021-02-08:
I added some doxygen to the header, could you please look at https://github.com/raspberrypi/pico-sdk/pull/101/commits/1141073b26fe4b26332493e4568c21a3faf94e54 and see if this covers what you wanted to know?

blippy, 2021-02-08:
Great! Yes, that covers it.

Many thanks.

closed 2021-02-10 by Wren6991