raspberrypi/pico-feedback #76: C/C++ Getting Started guide doesn't make clear how to quickly program the Pico over SWD

MichaelBell, 2021-02-09:
There's been a lot of discussion on the Discord and RPi forum about how to work around unplugging and plugging the USB cable while holding BOOTSEL to reprogram the Pico.  RPi Engineers are generally confused as to why users seem reluctant to use SWD.

I believe the problem is the Getting Started guide doesn't introduce the ability to program over SWD in a way that gives the user the ability to have a super simple way of building and running their program.  The thing I was looking for was to be able to do `make` then "flash" at a command line with no extra steps.

I hope these are some constructive suggestions on how to resolve that and encourage users to do the right thing:

- Split Chapter 5 into two chapters: "Programming over SWD" and "Debugging over SWD"
- "Programming over SWD" should have an introduction saying that programming over SWD allows you to reprogram the Pico without having to put it in BOOTSEL mode, and works even if the Pico has previously crashed.  It shouldn't talk about having to build in Debug, as that's not necessary if you just want to reprogram over SWD.
- "Programming over SWD" should then continue with what is currently section 5.2
- It should then have a short new section that tells you to run `openocd -f interface/raspberrypi-swd.cfg -f target/rp2040.cfg -c "program hello_world.elf verify reset exit"` to program the Pico.
- "Debugging over SWD" would then contain the rest of what is now chapter 5.  Splitting this into a separate chapter makes it clear that you can get a nice programming experience without needing to also install and work with gdb - you can go onto that once you need a debugger.

Extra thoughts:

- Maybe picotool could also gain a new command "swdload" which used openocd (either by telnet if it was already running, or by direct invocation if not) to load the program without having to use that rather long openocd command.
- Link forwards to the Programming over SWD section from 3.2.2 and 4.3 to hint to users becoming frustrated with the unplugging solution that there is a better way so they don't start trying to invent their own.

aallan, 2021-02-09:
This seems like a viable route to take. Thanks.

Wren6991, 2021-02-10:
Made a couple of changes in this direction in response to #78, but yes there is more to do.

> Split Chapter 5 into two chapters: "Programming over SWD" and "Debugging over SWD

Certainly an additional subheading, especially in conjunction with

> run `openocd -f interface/raspberrypi-swd.cfg -f target/rp2040.cfg -c "program hello_world.elf verify reset exit"`

Which should be called out explicitly. There was a TCL patch to make this possible (openocd gets confused about having multiple cores and one flash bank) which is currently on the rp2040 branch but not the picoprobe branch, so there needs to be a merge in one direction or the other first.

Maybe it would be clearer as a separate chapter as you say. I hadn't seen this issue when I responded to #78.

> Link forwards to the Programming over SWD section from 3.2.2

this was added

Wren6991, 2021-02-11:
Upstream we've now split chapter 5 into separate chapters on programming (with openocd) and debugging (openocd + gdb). It's still a bit sparse, and there is more work to do on the overall flow/navigability of the book, but it should make it into the next docs release.

aallan, 2021-02-11:
Thanks for the feedback, and we'll continue to poke at this, but for now I'm closing.

closed 2021-02-11 by aallan