raspberrypi/pico-feedback #77: Raspberry Pi Pico Python SDK Documentation Errata

Xeyler, 2021-02-09:
The document in question can be found here: https://datasheets.raspberrypi.org/pico/raspberry-pi-pico-python-sdk.pdf

Section 2.2 states: "REPL over UART is disabled by default."

Section 3.2 states: "USB serial is available from MicroPython, but the REPL is also available over UART0 by default."

These statements conflict with each other. A proper fix would be to test which statement is true and remove the false information. I'll test it soon.

panchalnikunj, 2021-02-13:
https://www.youtube.com/channel/UCA4j-J-TFW_9Dt4iTwIjtYQ

Once visit this channel for raspberry pi pico projects.

Xeyler, 2021-02-19:
Cursory testing would indicate that the UART is disabled by default, but before I can be sure, I'd like to test it with header pins soldered to my Pico. At the moment, I may just have a poor physical connection as I'm just holding wires to the UART pins.

If I get the same results after I've attached some headers, I'll try recompiling MicroPython with UART support and make sure I can connect to the REPL via serial.

aallan, 2021-02-19:
Fixed upstream. UART is disabled by default.

closed 2021-02-19 by aallan