raspberrypi/pico-feedback #78: Add pointers to practical flash programming info.

PiGraham, 2021-02-09:
It would be useful to reference in the contents that SWD is a convenient way to flash code to the PICO, avoiding the need to reboot with BOOTSEL pressed. It might also be a good idea to make that info a bit more prominent. It seems to be just this on Pg. 20:

> and load hello_world.elf into flash,
The following code illustrating commands to do that.

Maybe a heading "Easy Flash programming with SWD" or something like that.
And maybe a line to the effect: "it is not necessary to use the BOOTSEL button to program code into the flash if you connect an SWD interface."

Wren6991, 2021-02-10:
Thanks for raising this, it's important and this is a common theme we have seen. I added this section at the end of chapter 3:
[image]

Title of chapter 5 is changed from "Debugging with SWD" to "Flash Programming and Debug with SWD":
[image]

This new paragraph is at the top of chapter 5:

[image]

Wren6991, 2021-02-10:
We're also looking at ways of making the USB bootloader flow nicer (some way of resetting the board over USB if USB-serial is present in the binary) but this needs some more work and lots of testing across platforms.

closed 2021-02-10 by Wren6991