raspberrypi/pico-feedback #89: OUT command behavior

daveythacher, 2021-02-10:
I am reading over the datasheet for the PIO. I am not sure I understand the expected behavior of the OUT command from the OSR. 

I do not know for sure what the behavior of setting multiple shift bits is. The datasheet mentions "OUT PINS, 8 [1]" will shift a byte over two cycles. What is the timing of this? 
- serial to parallel (Load x bits into pin mapping on single cycle)
- skip over bits (Skip x - 1 bits and set all of pins to bit x value)
- shift bits against time divided clock (Shift all bits using state machine clock multiplex by number of bits)
- multicycle execution (Loop over command bit count times setting all pins to that bit value)

The examples provide an example of a single bit only, but not a multibit example. Multicycle can be achieved in assembly. Move can achieve serial to parallel. Skip over bits can be achieved with out to scratch pad, but is expensive. Shift bits against time divided clock would possibly put the performance in GHz range, which I am guessing is not correct.

The value of shift register is parallel to serial. I but I still do not know what the behavior against time is completely. Alternatives do exist for my personal application, but I am curious enough to ask. I could make argument for serial to parallel using pin mapper.

Other general feedback: 
I was thinking I had two OSR. There is a slight ambiguity of either direction, but this is realizable from the datasheet. There is really one fixed input and one fixed output. Both can perform left of right shift, not input/output. Little bit of a nit pick.

The examples have timing diagrams which are very useful, and I wish they were used more. I had to to a lot of work to understand the "[1]" of "OUT PINS, 8 [1]". Again I was able to figure this out from the datasheet. 

Side-set/delay encoding format would be nice to understand. I could probably figure this out if I wanted to, but I am not sure if there is a section for mentioning what values I populate there. However it is only a few combinations, I could just write every possible value into the assembler and read it back from hex. If I cared that much, there could be a rare usage of wanting to autogenerate instruction manually to pair up with exec function. There are hacks around this though even for this case.

lurch, 2021-02-11:
> Side-set/delay encoding format would be nice to understand.... I could just write every possible value into the assembler and read it back from hex.

Or you could have a look at the [pioasm source code](https://github.com/raspberrypi/pico-sdk/tree/master/tools/pioasm).

kilograham, 2021-02-11:
> > Side-set/delay encoding format would be nice to understand.... I could just write every possible value into the assembler and read it back from hex.
> 
> Or you could have a look at the [pioasm source code](https://github.com/raspberrypi/pico-sdk/tree/master/tools/pioasm).

better to read the datasheet which has the encodings

daveythacher, 2021-02-11:
Based on looking through the code and looking over the datasheet I found the answer. The PINCTRL_SIDESET_COUNT is what determines the encoding scheme not the instruction. This would have made attempting to brute force unsuccessful potentially. I am used to different assembly documentation.

Delay is the least significant bits, while side-set is the most significant bits. According to PINCTRL_SIDESET_COUNT partition. This is not obvious from code or datasheet. It is mentioned, but could be more straight forward. I am shocked it does not come out and be this blunt.

I would avoid turning anyone over to that code, it does somethings using a lot of tricks perhaps pointlessly. It flirted with bad coding practice. The code was difficult to track, as slick and compact as it was.

All of my other feedbacks are mentioned in the datasheet, but perhaps could be improved.

daveythacher, 2021-02-11:
This code could be improved:
https://github.com/raspberrypi/pico-sdk/blob/2d5789eca89658a7f0a01e2d6010c0f254605d72/tools/pioasm/python_output.cpp#L310-L318

uint delay = ((uint) inst >> 8u) & 0x1f;
uint side_set = delay >> sideset_bits_including_opt;
delay &= ((1u << (5 - sideset_bits_including_opt)) - 1u);

I am guessing that sideset_bits_including_opt is really the PINCTRL_SIDESET_COUNT value. This was not obvious to me first pass. This whole function is also duplicated. I still do not know what is happening in 312 and 313 to be honest, sideset_opt is a mystery to me. Trying to track some of that down took me on a wild chase of new messy C++ features.

aallan, 2021-02-12:
> All of my other feedbacks are mentioned in the datasheet

Closing.

closed 2021-02-12 by aallan

daveythacher, 2021-02-12:
> > All of my other feedbacks are mentioned in the datasheet
> 
> Closing.

I have four issues outline in the defect, three were none critical. My original issue still remains. Why was this closed?