raspberrypi/pico-feedback #90: Q: Behavior on Concurrent Instruction Insertion

soundpaint, 2021-02-12:
Sect. 3.2.2. of RP2040 datasheet says:

"Instructions written to a special configuration register ( SMx INSTR ) are immediately executed, momentarily interrupting
other execution."
and
"Instructions can be executed from a register, using the MOV EXEC instruction."

Now imagine, these two situations happen concurrently.  That is, for example, while a MOV EXEC instruction is still being decoded, there is also a DMA access to SMx INSTR writing another instruction.  Then there are _two_ pending instructions.

Now what happens?  According to the first statement, the instruction inserted via DMA would be executed, and only after that the instruction inserted by the MOV EXEC, and finally, ordinary execution would resume with the next program instruction.

Is my interpretation correct?  If not, maybe you could add some clarifying words to the documentation?

Thanks a lot,
Jürgen

Wren6991, 2021-02-17:
>  According to the first statement, the instruction inserted via DMA would be executed (discarding any delay cycles of the MOV EXEC instruction), and only after that the instruction inserted by the MOV EXEC (discarding any delay cycles of the instruction inserted via DMA), and finally, after any delay cycles of the instruction inserted by MOV EXEC, ordinary execution would resume with the next program instruction.

This is correct, except that MOV EXEC itself does not have any delay cycles, though its payload does (3.4.9):
[image]

Nor does SMx_INSTR (3.5.7):
[image]

It's done this way to avoid unnecessary complexity with multiple delay counters.

So in your hypothetical scenario, suppose the MOV EXEC takes place on cycle 0, and then the SMx_INSTR write takes place on cycle 1. The SMx_INSTR write will execute first, and the payload of MOV EXEC will be pushed back to cycle **2**.

This works as long as the instruction you write to SMx_INSTR does not stall, as there is a shared hardware resource used to latch that instruction if it stalls (3.5.7):

[image]

soundpaint, 2021-02-21:
Thank you!  I think I finally got it know based on your valuable answer! :-)

closed 2021-02-21 by soundpaint