raspberrypi/pico-feedback #91: ADC has very high DNL spikes and low-noise mode has more noise.

ferret-guy, 2021-02-13:
I have been working on testing some of the ADC parameters, as the datasheet is a little light. And it looks like there are significant DNL excursions indicating that the ADC is non-monotonic.

[image]

It also looks like the ADC performs better, across all the specifications when the converter is in PSM mode than PWM mode. [The RMS noise on the AVDD line is lower.](https://pico-adc.markomo.me/PSU-Noise/#rms-and-pk-pk-values)

ferret-guy, 2021-02-13:
Some additional looking shows that the 2047 code is missing on the 3 boards I tested.

aallan, 2021-02-15:
Just to cross-reference between the two, this is also being discussed [on the forums](https://www.raspberrypi.org/forums/viewtopic.php?p=1802658).

tony1tf, 2021-02-20:
> Just to cross-reference between the two, this is also being discussed [on the forums](https://www.raspberrypi.org/forums/viewtopic.php?p=1802658).

Thanks for the cross reference - I started that thread, so I am keen to get answers.

ferret-guy, 2021-02-20:
I assume the SAR DAC uses some kind of binary-weighted charge DAC, it looks like the binary-weighted capacitors for the 9 least significant bits are approximately 0.8% too large, and the 3rd least significant bit is approximately 0.85% too small.

Here is a plot of simulated vs measured DNL (click to see an HTML version that lets you zoom in):
[image]

Here is a table of results:

| Bit | Measured Binary Weight | Better Binary Weight | Delta  | Measured Size | Better Size |
|-----|------------------------|----------------------|--------|---------------|-------------|
| 1   | 1.0004                 | 1.0004               | 0.00%  | 5.00E-01      | 0.5002      |
| 2   | 1.0003                 | 1.0002               | 0.00%  | 2.50E-01      | 0.2501      |
| 3   | 0.9913                 | 0.9995               | 0.83%  | 1.24E-01      | 0.1249      |
| 4   | 1.0082                 | 0.9998               | -0.84% | 6.30E-02      | 0.0625      |
| 5   | 1.0114                 | 1.0029               | -0.85% | 3.16E-02      | 0.0313      |
| 6   | 1.0097                 | 1.0012               | -0.84% | 1.58E-02      | 0.0156      |
| 7   | 1.0075                 | 0.9991               | -0.84% | 7.87E-03      | 0.0078      |
| 8   | 1.0051                 | 0.9967               | -0.84% | 3.93E-03      | 0.0039      |
| 9   | 1.0210                 | 1.0124               | -0.86% | 1.99E-03      | 0.0020      |
| 10  | 0.9267                 | 0.9195               | -0.71% | 9.05E-04      | 0.0009      |
| 11  | 0.9160                 | 0.9090               | -0.70% | 4.47E-04      | 0.0004      |
| 12  | 0.9373                 | 0.9300               | -0.73% | 2.29E-04      | 0.0002      |

kevinjwalters, 2021-03-09:
I had a quick look at the RP2040 ADC too using CircuitPython, see appropriate graph in [Feather ADC comparison including 2.6V limited ESP32-S2](https://forums.adafruit.com/viewtopic.php?f=60&t=174020).

aallan, 2021-03-09:
Just an update, we're still investigating.

@jimbojr wrote [on the forums](https://www.raspberrypi.org/forums/viewtopic.php?f=144&t=299904&p=1833277#p1833277):
> A quick update on this - the issues are understood (we wanted to make sure all the data people are seeing/presenting was consistent with the known issues as well as make sure we completely understood the root cause(s) in the ADC design).
> 
> We are busy working on getting comprehensive characterisation data, which we will include in a datasheet update, along with a blog post.
> 
> In short, though, this is down to slight mismatch of some of the capacitors in the capacitive DAC, versus the simulation models.

We will hopefully have something definite to say in a week or two. Apologies for the delay.

aallan, 2021-03-16:
Just to let everyone know we're still working on this, we've even bought some new kit for the lab. Data is coming soon, and there will be a blog post — and probably an App Note-style PDF to back it up — in the next few weeks.

aallan, 2021-05-25:
A few weeks turned into a longer period of time due capacity issues. However we will be publishing a white paper on this issue shortly. Apologies for the delay.

lurch, 2021-05-25:
> A few weeks turned into a longer period of time due [to] capacity issues.

"this is down to slight mismatch of some of the capacitors in the capacitive DAC, versus the simulation models."

Oh the irony :rofl:

kevinjwalters, 2021-05-25:
It would be interesting to read about how these capacitors are created with the precise values (or these tweaked/trimmed?) needed or how variation is accomodated. I spotted the sawtooth pattern and assumed (out of ignorance) it was just down to the tolerances somewhere in the manufacturing process.

Even in the discrete world it's easy to forget there's a wide tolerance on capacitor values particularly because it's rarely marked on the capacitors nowadays.

A lot of the the capacitve-based charge-redistribution SA ADC explanations I've seen aren't particularly clear. If there's a plan to explain those I'd suggest road testing that with a normal person.

aallan, 2021-06-23:
We’ve completed characterisation of the RP2040 ADC and updated the [RP2040 Datasheet](https://datasheets.raspberrypi.org/rp2040/rp2040-datasheet.pdf). Please see the latest documentation release (tagged as V1.6) which was published today for further details, specifically sections 4.9.3 and 4.9.4. A chip errata (RP2040-E11) has been added to flag the issue for newcomers to the hardware.

closed 2021-06-23 by aallan