raspberrypi/pico-feedback #93: Possible crt0.S duplicate/useless CPUID wait_for_vector code?

mhinner, 2021-02-15:
Hi,

 this might be just my lack of understanding of the SDK code, however in bootrom I see call to wait_for_vector when CPUID!=0, this makes sense. https://github.com/raspberrypi/pico-bootrom/blob/master/bootrom/bootrom_rt0.S#L236

 However almost the same code is in SDK crt0.S, which in my opinion does not get ever executed (unless I provide vector of _start). Or am I missing something?

mhinner, 2021-02-15:
I also have a question why there are 2 registers named CPUID (I know it's mentioned in datasheet and understand the difference, just curious as it is in my opinion quite confusing ?).

Also I would prefer to have spinlock array defined in structs/sio.h such as :
#define sio_spinlock ((io_rw_32 *)(SIO_BASE+SIO_SPINLOCK0_OFFSET))  

It would be also great if you can add some use cases (or explain in more detail) SDK PDF chapters related to  __dmb, __isb, __mem_fence_acquire/release ; I have never programmed dualcore Cortex-M0 and I must say it's fun, just I am not 100% sure if/when to use dmb/isb in my mini SMP taskswitcher. I will learn more from ARM docs, this is just an idea.

Wren6991, 2021-02-16:
> However almost the same code is in SDK crt0.S, which in my opinion does not get ever executed

Did you see the comments about this in crt0.S, which confirm and explain what you said here?

```
.type _reset_handler,%function
.thumb_func
_reset_handler:
    // Only core 0 should run the C runtime startup code; core 1 is normally
    // sleeping in the bootrom at this point but check to be sure
    ldr r0, =(SIO_BASE + SIO_CPUID_OFFSET)
    ldr r0, [r0]
    cmp r0, #0
    bne hold_non_core0_in_bootrom
```

```
// ----------------------------------------------------------------------------
// If core 1 somehow gets into crt0 due to a spectacular VTOR mishap, we need to
// catch it and send back to the sleep-and-launch code in the bootrom. Shouldn't
// happen (it should sleep in the ROM until given an entry point via the
// cross-core FIFOs) but it's good to be defensive.

hold_non_core0_in_bootrom:
    ldr r0, = 'W' | ('V' << 8)
    bl rom_func_lookup
    bx r0
```

You can reach this point on core 1 by misconfiguring your debugger (e.g. putting both cores at the ELF entry point), so here we're spending 22 bytes in crt0 to make that scenario much less confusing.

mhinner, 2021-02-16:
> > However almost the same code is in SDK crt0.S, which in my opinion does not get ever executed
> Did you see the comments about this in crt0.S, which confirm and explain what you said here?

Sorry, no, looks like I have different crt0.S version, there is not even label hold_non_core0_in_bootrom (it's wait_for_vector in my crt0.S) . I should probably update the SDK more often :-), sorry for that.

lurch, 2021-02-17:
> I should probably update the SDK more often :-)

If you haven't already noticed, development of pico-sdk is happening on the [develop branch](https://github.com/raspberrypi/pico-sdk/tree/develop) (if you're looking for the latest&greatest version). And the bootrom sourcecode is [here](https://github.com/raspberrypi/pico-bootrom), although there's obviously no development being done on that :wink: 

Can this issue be closed?

closed 2021-05-25 by aallan