raspberrypi/pico-feedback #94: Typo on page 196 in RP2040 DATASHEET

JanHuygh, 2021-02-16:
On page 196 in the RP2040 DATASHEET the first reference to the SDK is printed twice.

aallan, 2021-02-16:
Issue transfered to `pico-feedback`

Wren6991, 2021-02-16:
Thanks, this is caused by a toolchain bug. Now fixed upstream, will appear in the next docs release.

closed 2021-02-16 by Wren6991

JamesH65, 2021-02-16:
How do you move an issue from one repo to another?

aallan, 2021-02-16:
> How do you move an issue from one repo to another?

Option is available at the bottom of the right-hand sidebar. Might only be possible because I'm a Github admin for the organisation though?

[image]