raspberrypi/pico-feedback #96: Additional  instructions for new gdb users.

PeterOGB, 2021-02-17:
Getting started with pico , Section 5.4:
No mention of how to stop your code running, or how to rebuild and reload new version.  These are fundamental setps  for new gdb users.

Suggest adding text to use Ctrl-C to stop a running program, and that "make" can be run from the gdb prompt.

Also add that a user command like "loadagain" can be used to replace the multiple gdb commands needed.
Add this to your ~/.gdbinint file
```
define loadagain
load
monitor reset init
continue
end
```

aallan, 2021-02-17:
The aim of the documentation is to cover Pico and RP2040. While we should potentially add a pointer to the GDB documentation, the point isn't to cover how to use GDB for new users. That's the responsibility of the GDB developers.

PeterOGB, 2021-02-17:
You've show how to install it, how to start it , how to conect it to openocd and  how to load and run a program.  Maybe you should take all that out because it's not your problem !   As it is you just leave them high and dry with no help as to what to do next !

aallan, 2021-02-17:
The "getting started" book is pretty long and covers a lot of fairly disparate topics. While I'm intending to add pointers to other documentation into it, I'm against adding large chunks of new content. Especially when it's something like gdb which is extensively documented elsewhere.

closed 2021-02-17 by aallan

PeterOGB, 2021-02-17:
Why do you think this is a "large chunk of content" ?  It's about 4 lines of text (if that)....

aallan, 2021-02-18:
We're at the point, possibly beyond it, where additional information in the getting started guide will just confuse people. I'm now actively looking to remove content from it, not add more content to it. While the next piece of documentation on the to do list is a 20 page(ish) summary datasheet for the RP2040, beyond that I'm thinking about an "Developing for Pico and RP2040" book. Additional content around gdb would be appropriate there.

PeterOGB, 2021-02-18:
OK, sounds a reasonable plan.  I still think leaving possibly new gdb users "hanging" at the end of the section should be addressed sooner rather than later.   I did find that I had to jump backwards a forwards through the getting started guide because I prefer to use "my favourite editor" plus  command line tools rather than VS Code.