raspberrypi/pico-feedback #100: Documentation: pico-sdk does not define panic

kiwih, 2021-02-18:
https://datasheets.raspberrypi.org/pico/raspberry-pi-pico-c-sdk.pdf frequently makes reference to 'panic'ing, but does not ever define exactly what a 'panic' is. 
Based on the definition in `pico/platform.h` I assume it is a non-returning printf of some kind, but it would be nice to have it spelled out somewhere.

aallan, 2021-05-25:
@kilograham What is the definition of "panic" used in the SDK book?

kilograham, 2021-10-21:
```
/*! \brief Displays a panic message and halts execution
 *  \ingroup pico_platform
 *
 * An attempt is made to output the message to all registered STDOUT drivers
 * after which this method executes a BKPT instruction.
 ```