raspberrypi/pico-feedback #98: DHT22 example page 241

NeilHDerby, 2021-02-18:
The example on page 241 indicates a DHT22 device and (unless i've missed something) is powering it from pin 36 which is a 3.3v supply. According to DHT22 datasheets it requires 3.5v - 5.5v! [I've tried one and with the 3.3v supply it will not work]

lurch, 2021-02-18:
Just for reference, the "upstream source" of this example is https://github.com/raspberrypi/pico-examples/tree/master/gpio/dht_sensor

JamesH65, 2021-02-18:
Adafruit says the DHT22 can run on 3v. https://learn.adafruit.com/dht

So it should be OK on the 3v3 line.

lurch, 2021-02-18:
...although ironically, a [later step](https://learn.adafruit.com/dht/connecting-to-a-dhtxx-sensor) of that same Adafruit tutorial also says "VCC - red wire Connect to 3.3 - 5V power. Sometime 3.3V power isn't enough in which case try 5V power." !
So maybe this is a product where multiple manufacturers make similar versions of the same product, so some sensors work with 3.3V and others don't? :man_shrugging:

daveythacher, 2021-02-18:
DHT11 is 3-5.5V: 
https://cdn-shop.adafruit.com/datasheets/DHT11-chinese.pdf
DHT22/AM2302 is 3.3-5.5V: 
https://cdn-shop.adafruit.com/datasheets/Digital+humidity+and+temperature+sensor+AM2302.pdf
https://cdn-shop.adafruit.com/datasheets/DHT22.pdf

closed 2021-05-25 by aallan